lemmingDev/ESP32-BLE-Gamepad #254: I have a problem with library files

Auyp, 2025-01-08:
When I try to use the example inside the library or any code I write and then try to compile it with a ESP32 board, this error appears.
[image]
I tried a lot but I did not know the reason for this error even though I am using the latest version of the library with the latest version of the NimBLE-arduino library 

This is one of the ones I tried :

```
#include <Arduino.h>
#include <BleGamepad.h>

BleGamepad bleGamepad;

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting BLE work!");
    bleGamepad.begin();
   
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        Serial.println("Press buttons 5, 16 and start. Move all enabled axes to max. Set DPAD (hat 1) to down right.");
        bleGamepad.press(BUTTON_5);
        bleGamepad.press(BUTTON_16);
        bleGamepad.pressStart();
        bleGamepad.setAxes(32767, 32767, 32767, 32767, 32767, 32767, 32767, 32767);
        bleGamepad.setHat1(HAT_DOWN_RIGHT);
        delay(500);

        Serial.println("Release button 5 and start. Move all axes to min. Set DPAD (hat 1) to centred.");
        bleGamepad.release(BUTTON_5);
        bleGamepad.releaseStart();
        bleGamepad.setHat1(HAT_CENTERED);
        bleGamepad.setAxes(0, 0, 0, 0, 0, 0, 0, 0);
        delay(500);
    }
}

```

LeeNX, 2025-01-08:
Have you installed ```ESP32-BLE-Gamepad``` and ```NimBLE-Arduino```  libraries in your Arduino IDE? 

or you can try PlatformIO and put into your ```platformio.ini``` for the project
```ini
lib_deps =
        lemmingdev/ESP32-BLE-Gamepad@^0.6.1
        h2zero/NimBLE-Arduino@^2.1.3
```

Auyp, 2025-01-08:
@LeeNX Yes I installed the libraries in Arduino IDE. I will try the PlatformIO today

LeeNX, 2025-01-08:
Have you tried to confirming the files exists in your Arduino IDE Library folder, on my Mac in my Documents folder, looks something like
```bash
leet@CLees-MacBookPro11-2 ESP32-BLE-Gamepad % pwd
/Users/leet/Documents/Arduino/libraries/ESP32-BLE-Gamepad
leet@CLees-MacBookPro11-2 ESP32-BLE-Gamepad % ls -la BleGamepad.h
-rw-r--r--  1 leet  staff  4035 Dec 16 14:36 BleGamepad.h
```
Might be a path issue, but I have not run into this, I don't have any other ideas to offer.

Auyp, 2025-01-08:
@LeeNX  Yes, I checked that it exists.

Auyp, 2025-01-08:
I solved the problem. It turned out that it was due to the presence of Arabic words in the path. 

C:\Users\ethgd\OneDrive\المستندات\

closed 2025-01-08 by Auyp