lemmingDev/ESP32-BLE-Gamepad #255: [Feature Request] Unit testing in CI

LeeNX, 2025-01-09:
Please can we get some unit testing in CI.  
  
This issue I think is more a request in insight and ideas on how we could possible do unit testing in CI.  
  
I don't think we could emulate an ESP32 and/or bluetooth connections and device, thou if somebody else knows otherwise, that would be awesome.

Next best thing I can think of, is a few devices setup that can be flashed with some of the sketches that run some tests.

lemmingDev, 2025-02-01:
Yeah - not sure how this would be done

LeeNX, 2025-02-02:
I don't believe there is a way to emulate this in GitHub action in a runner, but we could do a self-hosted runner, which could use two ESP32 devices, one with Bluepad32 as a host and then flash the latest testing sketch onto the other ESP32 device.  
  
This would be a great place to also do the [Latency Measurements](https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/148). Test that all endpoints (buttons/sliders/joysticks/etc) work as expected.  
  
Both PlatformIO and Arduino cli should be able to do the build and flash to the device. Have not figure out how to do the Bluepad32 flashing (it's a custom board setup in Arduino), but maybe we don't need to change that, it might just stay static.

Also let people setup test rigs of their own, so that we can reduce the number of variables when trouble shooting issues.

LeeNX, 2026-08-28:
I have finally got a chance to work on this feature, basically building a CI for hardware-in-the-loop (HIL), with at least two different ESP32 MCUs (ESP32/ESP32-C3). 

Building a basic test of tests that can check that some of the features of ESP32-BLE-Gamepad are working.

Let's see where this goes.

If anybody is interested, drop in note in the issue and I will update more often.

LeeNX, 2026-08-30:
Hey @lemmingDev , please can you look over the following and give me you feedback.

I have made a repo - [ESP32-BLE-Gamepad-HIL](https://github.com/LeeNX/ESP32-BLE-Gamepad-HIL), for the hardware-in-the-loop(HIL) parts, as I did not think this should be hosted in the ```ESP32-BLE-Gamepad``` repo, but maybe it should. I am happy if you would like to create this repo in your nameSpace and then I can push to that, might be better, totally your call?

PR https://github.com/lemmingDev/ESP32-BLE-Gamepad/pull/347 is in place, runs locally, working out the CI and some remote access via TailScale.

Feedback welcome.

lemmingDev, 2026-08-30:
Happy to put it here if you like - totally up to you
You've got (or will have) access anyway
Just let me know if i need to do anything

Pull #347 and anything else you need to

LeeNX, 2026-08-30:
So many moving parts, but I think it got the PoC part working - https://github.com/LeeNX/ESP32-BLE-Gamepad-HIL/actions/runs/33314057913/job/99264550284

Now I wonder if it's been hardening enough from a security point of view?

Oh, and now I wonder if I was not a little flash happy, I think it flashing about 6 times per run. 

Still needs a little more testing, but I think I have all the basic parts in place, what do you think?

Does it cover enough testing? Anything you would like added or changed?

LeeNX, 2026-08-30:
I think you need to be logged in to GitHub, but - https://github.com/LeeNX/ESP32-BLE-Gamepad-HIL/actions/runs/33314057913/attempts/1#summary-99264550284

Lots of green and now red, a few skips, but I think that is one script trying to cover all. I need to bump some actions, but that should be easy.

lemmingDev, 2026-08-30:
LeeNX, this is an absolutely phenomenal piece of work. What you've built is genuinely impressive — a complete HIL test rig with builder/tester split, robust serial protocol, BlueZ pairing that actually works, golden-file HID descriptor comparison, and real latency measurements that surface actual findings (the 79-button Linux limit, the reversed hat fields, the Feature Report off-by-one). These are the kinds of things that only show up with real hardware testing, and they're incredibly valuable for anyone using the library.

The architecture is really well thought out:

- The **builder/tester split** means the test runner can be a Pi that can't build firmware, while the builder is any machine with PlatformIO
- The **serial command protocol** is clean and extensible — easy to add new commands
- The **profile system** makes it trivial to test different HID descriptor configurations
- The **Tailscale CI story** is slick — no self-hosted runner needed, just an SSH target on the tailnet
- The **bond state tracking** with `{mac, profile}` in `state.json` is a nice touch for automatic re-pairing

The findings are already actionable:
- The ~79-button Linux limit is critical knowledge for anyone designing a gamepad
- The hat field reversal explains behavior I've seen but never pinned down
- The 18.6ms median latency is a solid baseline
- The burst overflow finding validates the "don't sendReport() faster than BLE can transmit" guidance

---

**On multi-mode support:** I'm working on the `multi-gamepad` branch of the library, which adds SInput (SDL3 native) and XInput (Xbox emulation) modes alongside the existing Generic mode. I think the HIL rig would be the perfect place to validate these modes. Here's what would be needed:

### Firmware Changes

**New profiles in `hil_profile.h`:**

```cpp
// SInput mode (25 buttons, 6 axes, 1 hat, fixed VID 0x2E8A / PID 0x10C6)
PROFILE(sinput_default, 25, 1, 6, 0, 0x2E8A, 0x10C6, "SInput: 25 buttons, sticks, triggers")

// XInput mode (11 buttons, 6 axes, 1 hat, fixed VID 0x045E / PID 0x02FD)
PROFILE(xinput_ones,    11, 1, 6, 0, 0x045E, 0x02FD, "XInput One S: 11 Xbox buttons")
PROFILE(xinput_seriesx, 11, 1, 6, 0, 0x045E, 0x0B13, "XInput Series X: Share button")
```

**New serial commands for mode-specific features:**

| Command | Purpose |
|---|---|
| `MODE?` | Returns current mode (Generic/SInput/XInput/XInputSeriesX) |
| `RUMBLE?` | Check if rumble command received (SInput haptic / XInput PID Set Effect) |
| `PLAYERLED <n>` | Set player LED index (SInput only) |
| `RGB <r> <g> <b>` | Set RGB LED color (SInput only) |
| `TOUCHPAD <pad> <x> <y> <p>` | Set touchpad position/pressure (SInput only) |
| `IMU <ax> <ay> <az> <gx> <gy> <gz>` | Set accelerometer/gyro data (SInput only) |

The firmware would need a `GamepadMode` parameter passed via the profile or a serial command, since `begin()` sets the mode once and the HID descriptor + BLE characteristics are mode-specific.

### Host Test Changes

**New test modules:**

| Module | Purpose |
|---|---|
| `host/tests/test_sinput.py` | SInput protocol validation |
| `host/tests/test_xinput.py` | XInput protocol validation |
| `host/hil/sinput.py` | SInput helper (Features response parsing, usage mask validation) |
| `host/hil/xinput.py` | XInput helper (Xbox button bitmask validation) |

**Key test scenarios:**

1. **SInput 25-button mapping** — Press each BUTTON_1..25, verify distinct evdev key events in the correct range
2. **SInput Features response** — Validate capability bits, usage masks 0-3, gamepad type, face style
3. **SInput touchpad offsets** — Send touchpad data, verify correct byte positions (we recently fixed an off-by-one: offsets should be 34/36/38/40/42/44, not 35/37/39/41/43/45)
4. **SInput rumble/LED/RGB** — Send output commands, verify reception via serial queries
5. **XInput 11-button mapping** — Press each Xbox button, verify correct mapping to Xbox button bitmasks
6. **XInput rumble** — Send PID Set Effect Report, verify strong/weak motor values
7. **VID/PID warnings** — Override VID/PID in SInput/XInput mode, verify Serial warning is printed

### CI Matrix

Add the new profiles to the build matrix in `.github/workflows/hil.yml`:

```yaml
strategy:
  matrix:
    profile: [default, signed-axes, specials, minimal, maxbtn, reports,
              sinput_default, xinput_ones, xinput_seriesx]
```

### Estimated Effort

- **New profiles**: ~30 minutes (just `#define PROFILE(...)` entries)
- **Firmware serial commands**: ~2-3 hours (mode-aware command handlers)
- **SInput host tests**: ~4-6 hours (helper module + test cases)
- **XInput host tests**: ~3-4 hours (helper module + test cases)
- **CI matrix update**: ~15 minutes

**Total: ~1-2 days** for someone familiar with both codebases. Alternatively, I could get AI to do the grunt work for the PR in around an hour — generate the profiles, serial commands, test scaffolding, and CI matrix — and then you could add your polish and domain expertise on top.

---

I'd be happy to submit a PR with these changes if that would be useful. The HIL rig is exactly what the multi-gamepad branch needs for validation — compile-only CI catches build errors, but only real hardware testing can verify that `press(BUTTON_13)` actually produces the correct Capture/Share button event on a host, or that touchpad data arrives at the right byte offsets.

Thanks again for building this. It's a huge contribution to the project.

lemmingDev, 2026-08-30:
Looks good (from a layman's standpoint like mine, with a bit of AI analysis thrown in for good measure) to me

LeeNX, 2026-08-31:
Thanks @lemmingDev , I can't say I can take all the credit, got the idea for TailScale from [Gina Häußge](https://github.com/foosel) and the CI she explained for her famed project [OctoPrint](https://github.com/OctoPrint/OctoPrint)

I am happy to take PR's in my repo and I can setup testing for the other modes, my first two worries are, how quick are we going to hit the flash ware limits in these devices. I was already thinking of merging some of the profiles, so we have less flashing, but we can worry about that a little further down the line.

The other is security, I need to run another look over and see if we have any possible issues, before we just open it up to anybody. Maybe we do basic testing on PRs, unless the PR is targeting something like bench vs new feature or something. And complete run on releases only.

Looking to see if I have any other MCU's in my pile of items I have ordered but not got around to putting in place.

Let me setup a release process in place, so that we can work off some fixed versions, but I am happy with any changes and improvements, as long as we can keep security in mind. 

I also need to put this R-Pi into a box with better cooling and not open to the air and dust. I think I will add some notes in my HIL repo.

Looking forward to your feedback.

I also need to get this working in ESP32-BLE-Gamepad repo, so that it's called and run.

LeeNX, 2026-10-12:
Release and summary of testing, which includes some basic SInput - https://github.com/LeeNX/ESP32-BLE-Gamepad-HIL/actions/runs/37648035508

I still want to see what we can include with the above suggestions, but for now I think this is working pretty well and without any further feedback, I am going to let it soak and see what else we turn on.