lemmingDev/ESP32-BLE-Gamepad #259: Issues and Feature requests

DrSimgear, 2025-01-22:
There is this #include <driver/adc.h> in https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/d43027a7e60e1616fa09afee43d07478033c6dfa/BleGamepad.cpp#L7
[picture](https://cdn.discordapp.com/attachments/863150247617363998/1331389611531964477/image.png?ex=679170b3&is=67901f33&hm=d2848cedff28520c9a58f9ab7c9ca6c2d39d8f328c7ab2cec4b1cba194429fc8&)
I'm not sure why its there but I was getting this warning so I removed it and warning went away.

Second Issue;
Throttle axis is int16 but it should be uint16. When I set 65535 and 0 as max min, it doesn't really work the way I intend since the max is 32767. I also use the Joystick_ESP32S2 library in combination with it for usb gamepad so I switch between usb and ble on the fly. On that using 65535 and 0 works perfectly well but if I connect through ble then the numbers go into negative because anything above 32767 loops around to -32767. 

Third; 
I would like to be able to disconnect clients and start/stop advertising ble. There is functions in nimble arduino library to do this but I couldn't figure it out how to add them to my arduino program because I couldn't get the syntax right. First I tried to turn off ble all together but when I turned it back on I would get overflow because there is a procedure you have to follow to turn off ble properly, someone posted [how](https://www.esp32.com/viewtopic.php?t=32972) for esp idf but I couldn't get the syntax right for it in arduino ide. Basically I would like for esp32 ble connection to cut out when I plug in the usb into a computer at which point the usb gamepad takes over or vice versa, since they look like the same device with the same vid and pid, etc. games start getting inputs from the usb without having to rebind etc.

LeeNX, 2025-01-22:
Thanks @DrSimgear for the report. Some suggestions to make tracking and offering assistants easier, would be break your issue up into small, clear issues, each of there own. You could link them in the description or comments if they are related. Just makes things little easier for everybody. Also providing a link to your code or snippets showing the problem would take a lot of guess work out for people wanting to help.  
  
First issue:
might be worth removing. I don't see where any ADC stuff is happening directly in the library. Why don't you make a PR, then we can test. This will get you GitHub street cred. Somebody else might do it, if you don't know how, but maybe you would like to try and we can help you along.

Second issue: 
I have not used axis other than in the test sketchers, so can't give you a clear answer, but hopefully I can offer some suggestions. I think the HID description is for two byte value. Setting the minimum value to zero, might help?
As outline in ```FlightControllerTest.ino```
https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/d43027a7e60e1616fa09afee43d07478033c6dfa/examples/FlightControllerTest/FlightControllerTest.ino#L41-L46
  
Third:
I am guessing you looking at turn off bluetooth in your sketch vs disconnecting the currently connected client. This is where splitting this up into multiple issues would be better, because now discussing this item does not really relate to the other items.
I have a ToDo item to ban a client from connecting or just disconnecting, something like
```
void disconnectClient() {
    NimBLEServer* server = NimBLEDevice::getServer();
    if (server) {
        // Iterate through connected peer devices
        for (uint16_t connHandle : server->getPeerDevices()) {
            server->disconnect(connHandle); // Disconnect the client
            Serial.printf("Disconnected client with connHandle: %d\n", connHandle);
        }
    } else {
        Serial.println("No server instance available.");
    }
}

void loop() {
    // For demonstration: disconnect the client when a button is pressed
    if (bleGamepad.isConnected()) {
        Serial.println("BLE Client connected");
        
        // Add your gamepad logic here
        
        // Simulate a disconnect action
        if (digitalRead(0) == LOW) { // Assuming a button is wired to pin 0
            disconnectClient();
        }
    }

    delay(100); // Adjust as needed
}
```
This just compiles, I have not tested further.

Let us know how it does.

DrSimgear, 2025-01-24:
Hi, Thanks for the reply @LeeNX . I made a pr, let me know if it's good to go. Sorry I shouldn't have combined multiple issues, I didn't want to add spam too much. For the third issue, I will give the code a try on the weekend. I think I will need to add the advertising part to it too because I want windows to reconnect back when I unplug usb. 

Second issue: https://github.com/DrSimgear/OH58-Kiowa-Collective
I created a repository for my code to make things easier. Basically throttle in BLE Gamepad is a signed 16bit value. Which goes from -32767 to +32767. Even if I set it to 0 and 65535 which is a unsigned 16bit value, if my throttle goes above +32767 then it will loop back to -32767 in windows.

DrSimgear, 2025-01-25:
@lemmingDev I updated to newest version and flashed same firmware but axis output from ble is stuck at 0. It is sending button states though and I can see axis position change in serial monitor.

lemmingDev, 2025-01-25:
ooh - should've tested myself - I see no axes are working now after the recent release

I've just tested myself, and there didn't seem to be any issues in the first place in all of the built-in examples for my library before I updated it trying to fix an issue (that perhaps doesn't exist?)

I'll release a new version reverting back to the way it was and then might check your controller code for issues

Thanks

lemmingDev, 2025-01-25:
Here is an updated driving controller test sketch where you can set the axes min to either a zero or negative value, and it works perfectly across the entire range

I cannot find any issue whatsoever

Please test and confirm the same with [updated v0.6.8 release](https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.6.8) 

```
/*
 * Driving controller test
 */

#include <Arduino.h>
#include <BleGamepad.h>

#define numOfButtons 10
#define numOfHatSwitches 0
#define enableX false
#define enableY false
#define enableZ false
#define enableRX false
#define enableRY false
#define enableRZ false
#define enableSlider1 false
#define enableSlider2 false
#define enableRudder false
#define enableThrottle false
#define enableAccelerator true
#define enableBrake true
#define enableSteering true

//int16_t simMin = 0x8000;      // -32767 --> Some non-Windows operating systems and web based gamepad testers don't like min axis set below 0, so 0 is set by default
//int16_t axesCenter = 0x00;      
int16_t simMin = 0x00;        // Set simulation minimum axes to zero.
int16_t axesCenter = 0x3FFF;
int16_t simMax = 0x7FFF;        // 32767 --> int16_t - 16 bit signed integer - Can be in decimal or hexadecimal
int16_t stepAmount = 0xFF;
uint16_t delayAmount = 25;

BleGamepad bleGamepad("BLE Driving Controller", "lemmingDev", 100);

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting BLE work!");

    // Setup controller with 10 buttons, accelerator, brake and steering
    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setAutoReport(false);
    bleGamepadConfig.setControllerType(CONTROLLER_TYPE_GAMEPAD); // CONTROLLER_TYPE_JOYSTICK, CONTROLLER_TYPE_GAMEPAD (DEFAULT), CONTROLLER_TYPE_MULTI_AXIS
    bleGamepadConfig.setButtonCount(numOfButtons);
    bleGamepadConfig.setWhichAxes(enableX, enableY, enableZ, enableRX, enableRY, enableRZ, enableSlider1, enableSlider2);      // Can also be done per-axis individually. All are true by default
    bleGamepadConfig.setWhichSimulationControls(enableRudder, enableThrottle, enableAccelerator, enableBrake, enableSteering); // Can also be done per-control individually. All are false by default
    bleGamepadConfig.setHatSwitchCount(numOfHatSwitches);                                                                      // 1 by default
    bleGamepadConfig.setSimulationMin(simMin);
    bleGamepadConfig.setSimulationMax(simMax);
    
    bleGamepad.begin(&bleGamepadConfig);
    // changing bleGamepadConfig after the begin function has no effect, unless you call the begin function again

    // Set steering to center
    bleGamepad.setSteering(axesCenter);

    // Set brake and accelerator to min
    bleGamepad.setBrake(simMin);
    bleGamepad.setAccelerator(simMax);

    bleGamepad.sendReport();
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        Serial.println("Press all buttons one by one");
        for (int i = 1; i <= numOfButtons; i += 1)
        {
            bleGamepad.press(i);
            bleGamepad.sendReport();
            delay(100);
            bleGamepad.release(i);
            bleGamepad.sendReport();
            delay(25);
        }

        Serial.println("Move steering from center to min");
        for (int i = axesCenter; i > simMin; i -= stepAmount)
        {
            bleGamepad.setSteering(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }

        Serial.println("Move steering from min to max");
        for (int i = simMin; i < simMax; i += stepAmount)
        {
            bleGamepad.setSteering(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }

        Serial.println("Move steering from max to center");
        for (int i = simMax; i > axesCenter; i -= stepAmount)
        {
            bleGamepad.setSteering(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }
        bleGamepad.setSteering(axesCenter);
        bleGamepad.sendReport();

        Serial.println("Move accelerator from min to max");
        // Axis is reversed, so swap min <--> max
        for (int i = simMax; i > simMin; i -= stepAmount)
        {
            bleGamepad.setAccelerator(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }

        Serial.println("Move accelerator from max to min");
        // Axis is reversed, so swap min <--> max
        for (int i = simMin; i < simMax; i += stepAmount)
        {
            bleGamepad.setAccelerator(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }
        bleGamepad.setAccelerator(simMax);
        bleGamepad.sendReport();

        Serial.println("Move brake from min to max");
        for (int i = simMin; i < simMax; i += stepAmount)
        {
            bleGamepad.setBrake(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }
        
        Serial.println("Move brake from max to min");
        for (int i = simMax; i > simMin; i -= stepAmount)
        {
            bleGamepad.setBrake(i);
            bleGamepad.sendReport();
            delay(delayAmount);
        }
        bleGamepad.setBrake(simMin);
        bleGamepad.sendReport();
    }
}
```

lemmingDev, 2025-01-25:
Oh, and don't use anything over 0x7FFF / 32767 in the axes max, as over that goes into negatives values when you assign them to the USB HID descriptor

It's standard int16_t behaviour

Eg, 

```
  Serial.println((int16_t)0x7FFE);
  Serial.println((int16_t)0x7FFF);
  Serial.println((int16_t)0x8000);
  Serial.println((int16_t)0x8001);
  Serial.println((int16_t)0xFFFE);
  Serial.println((int16_t)0xFFFF);
```
 gives you
```
32766
32767
-32768
-32767
-2
-1
```

lemmingDev, 2025-01-25:
Your 65535 is actually 0xFFFF which is -1

lemmingDev, 2025-01-25:
I just checked that library

You should be able to set the 2 lines from

```
#define JOYSTICK_AXIS_MAXIMUM 65535
#define JOYSTICK_SIMULATOR_MAXIMUM 65535
```
to 
```
#define JOYSTICK_AXIS_MAXIMUM 32767
#define JOYSTICK_SIMULATOR_MAXIMUM 32767
```
and then use 32767 as the max and 0 as the min when setting up my library, and then both libraries should be able to coexist fine in your sketch

DrSimgear, 2025-01-26:
Thanks for the help. Since last update bluetooth axis started updating again. 

 With my use any more than 10bit is overkill and I did set the other library to 4095 and the output goes to 4095 max but the range is still 65535 so I need to dig deeper. Not sure why they set it like that and not give users a easy way to change it using the api. I did manage to modify the ble library to report 0 to 65535 in windows with succes, maybe giving users a way to easily switch between uint and int would be good. A long time ago I remember trying to calibrate some axis in a game and it was expecting axis to go into negative but because I set it to 0 - 65535  the calibration could never complete. The opposite may happen too for example if a game expects throttle values as unsigned then something might break.

I noticed in your library you fixed the if statements to limit the values to what the user set. You could change that to constrain function for extra performance and readability, but I dont think its needed in the first place because users can add constraint function in their own code for when it's needed just like this "x = constrain(y, 0, 4095);".

lemmingDev, 2025-01-27:
Hi. Thanks for following it up

Another easier fix in your sketch would have been to just divide the 0 - 65535 value by 2 before setting the BLE axis vs the full value for setting the USB library axis

I did, for a short time while looking at the issue you posted (before reading it fully), do a check to constrain values between only possible user set min and max values, but ultimately got rid of it as I realised there was no real issue with the library and was easier just to revert to a previous version of the file. I might put that check back at some stage

As for an easy way to allow larger axes values, feel free to submit a pull request for it

Anyway, have the issues in your initial post been resolved?

DrSimgear, 2025-01-28:
Yes, the first and second issue are resolved. Haven't had a chance to mess with the third one but I can close this for now if you would like.

LeeNX, 2025-01-28:
Sounds like progress all round, thanks. @DrSimgear let us know if you come right with the BLE disconnect/de-advertise. I think that would be nice to add to my own project for the perfect DIY gamepad.

lemmingDev, 2025-01-28:
> Sounds like progress all round, thanks. [@DrSimgear](https://github.com/DrSimgear) let us know if you come right with the BLE disconnect/de-advertise. I think that would be nice to add to my own project for the perfect DIY gamepad.

I am playing with that now

lemmingDev, 2025-01-28:
So, I was able to get it working where:

1. My Android phone was connected to it
2. Press a button and have it delete all bonds and automatically restart board
3. Android did not automatically connect back to it 
4. Windows 11 PC was able to connect to it fine

Only issue is that I couldn't do it in reverse. Windows briefly connects to the board for a short time and prevents other devices from seeing it. Need to unpair/remove

I'll test further

LeeNX, 2025-01-28:
Thanks @lemmingDev . I think the idea thou is a button to enable/disable BLE. Which would mean disconnect all hosts(bonds) and disable advertising until the enable BLE happens, either via button or event, like USB been disconnected. When advertising is enabled, I would then expect hosts to reconnect, if they are setup to re-connect. I think I have seen an option in some OS, not sure which one, that you can disable auto-reconnect.   
  
The Windows reconnect and stopping other connections, I have been thinking about banning a current connection, like in my feature request - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/184 ... I just don't know how we could delete bans without some sort of interface.

lemmingDev, 2025-01-29:
Ok - so issue now totally resolved
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.6.9

New version allows 3 methods of dealing with this

Most gentle one is bleGamepad.enterPairingMode() which repeatedly disconnects from currently connected device until a new device is paired
If you want a more permanent solution, instead use bleGamepad.deleteBond() which will delete the bond for the currently connected client and allow other clients to connect to it

The most drastic is bleGamepad.deletAllBonds() which deletes all bonds from the gamepad

After deleting bonds, it is best to unpair them at some stage from the client device such as your phone or PC otherwise the gamepad may briefly connect to them while searching

lemmingDev, 2025-01-30:
Please test and report back @DrSimgear

DrSimgear, 2025-01-30:
Thank You, I will try. I am a little confused why the esp needs to be reseted after disconnect as that would be the same as just resetting it. Does using disconnect(connHandle) and then stop advertising not work to prevent pc from connecting back? 
https://h2zero.github.io/NimBLE-Arduino/class_nim_b_l_e_advertising.html

lemmingDev, 2025-01-30:
Correct - It doesn't need to be reset at all.

Advertising just needs to be restarted

I just added it as an optional feature that you can happily ignore as the default is false

lemmingDev, 2025-02-01:
Will close if all good

DrSimgear, 2025-02-01:
Thanks for the help guys. I will test and maybe fork if I figure something out.

closed 2025-02-01 by DrSimgear