lemmingDev/ESP32-BLE-Gamepad #261: Issue 4 encoders and 20 buttons

Gearhead-Ceniza, 2025-01-26:
``Good afternoon fellas

I am having problems with my code/setup. I'm building a sin wheel and the problem I've got is a little strange. Via serial monitor the code seems to be working fine, every button pressed gets a reading. From 0 to 19 every single button pressed works and the encoders read pulses from 20 to 28. The problem is that those numbers doesn't correspond to the pad in the tests.

Using joysticktest_092 de correlation pressing the buttons between serial monitor and the reading in software is the next.

SerialMonitor | joysticktest_092
        1             |        1
        2             |        2
        3             |        4
        4             |        8
        5             |        16
        6             |         32
    7|64
etc.
I've modified gamepad.h

```c++
private:
    uint8_t _buttons[64]; // 8 bits x 16 --> 128 bits
``` 
instead of
```c++
private:
    uint8_t _buttons[16]; // 8 bits x 16 --> 128 bits
``` 

And also BleGamepadConfiguration.cpp

```c++
#include "BleGamepadConfiguration.h"

BleGamepadConfiguration::BleGamepadConfiguration() : _controllerType(CONTROLLER_TYPE_GAMEPAD),
                                                     _autoReport(true),
                                                     _hidReportId(3),
                                                     _buttonCount(64),
                                                     _hatSwitchCount(0),
``` 
if I'm not wrong, that is needed to show up to 64 buttons instead of the 16 buttons by default. 
Here is my code. I hope someone finds what's wrong. I'm not used to write any code and this is just a modified example that I´ve found here https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/40  

```c++
#include <ESP32Encoder.h>  // https://github.com/madhephaestus/ESP32Encoder/
#include <Keypad.h>        // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>    // https://github.com/MagnusThome/ESP32-BLE-Gamepad

BleGamepad bleGamepad("SIM_WHEEL", "gearhead", 100);

////////////////////// BUTTON MATRIX //////////////////////
#define ROWS 5
#define COLS 4
uint8_t rowPins[ROWS] = { 13, 14, 15, 16, 33 };
uint8_t colPins[COLS] = { 17, 18, 19, 21 };
uint8_t keymap[ROWS][COLS] = {
  { 0, 1, 2, 3 },
  { 4, 5, 6, 7 },
  { 8, 9, 10, 11 },
  { 12, 13, 14, 15 },
  { 16, 17, 18, 19 }
};
Keypad customKeypad = Keypad(makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

//////////// ROTARY ENCODERS (WITH PUSH SWITCHES) ////////////
#define MAXENC 4
uint8_t uppPin[MAXENC] = { 22, 25, 27, 04 };
uint8_t dwnPin[MAXENC] = { 23, 26, 32, 05 };
uint8_t encoderUpp[MAXENC] = { 20, 22, 24, 26 };
uint8_t encoderDwn[MAXENC] = { 21, 23, 25, 27 };
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = { 0, 0, 0, 0 };
int32_t prevenccntr[MAXENC] = { 0, 0, 0, 0 };
bool prevprs[MAXENC] = { 0, 0, 0, 0 };
#define HOLDOFFTIME 150  // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

void KeypadUpdate() {
  customKeypad.getKeys();

  for (int i = 0; i < LIST_MAX; i++)  // Scan the whole key list.
  {
    if (customKeypad.key[i].stateChanged)  // Only find keys that have changed state.
    {
      uint8_t keystate = customKeypad.key[i].kstate;

      if (keystate == PRESSED) { pressKey(int(customKeypad.key[i].kchar)); }
      if (keystate == RELEASED) { releaseKey(int(customKeypad.key[i].kchar)); }
    }
  }
}

void sendKey(uint8_t key) {
  uint32_t gamepadbutton = pow(2, key);  // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("pulse\t");
  Serial.println(key);
  if (bleGamepad.isConnected()) {
    bleGamepad.press(gamepadbutton);
    delay(150);
    bleGamepad.release(gamepadbutton);
  }
}

void pressKey(uint8_t key) {
  uint32_t gamepadbutton = pow(2, key);  // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("hold\t");
  Serial.println(key);

  if (bleGamepad.isConnected()) {
    bleGamepad.press(gamepadbutton);
  }
}

void releaseKey(uint8_t key) {
  uint32_t gamepadbutton = pow(2, key);  // CONVERT TO THE BINARY MAPPING GAMEPAD KEYS USE
  Serial.print("release\t");
  Serial.println(key);

  if (bleGamepad.isConnected()) {
    bleGamepad.release(gamepadbutton);
  }
}

void setup() {
  Serial.begin(115200);

  for (uint8_t i = 0; i < MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachFullQuad(dwnPin[i], uppPin[i]);
  }

  //customKeypad.setHoldTime(1);
  bleGamepad.begin();
  Serial.println("Booted!");
}

void loop() {
  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i = 0; i < MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();

    if (cntr != prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr > prevenccntr[i]) { sendKey(encoderUpp[i]); }
        if (cntr < prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i] == 0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      } else if (now - holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
  }

  KeypadUpdate();

  delay(10);
}
``` 

Thanks.

Ceniza

LeeNX, 2025-01-27:
Hey @Gearhead-Ceniza , ran into a YT video over the weekend which a 20 button setup, which sounds very similar to what you looking to do.  
  
My first question is, instead of making changes to the library, why not just set the number of buttons you want and let the setup function build all the needed bits.  
  
A good example sketch to start with is [DrivingControllerTest](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/DrivingControllerTest/DrivingControllerTest.ino). 
  
Should be able to update - https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/78323097d3c0aed509852f1ff91ede30e0882808/examples/DrivingControllerTest/DrivingControllerTest.ino#L8

to
```
#define numOfButtons 20
```
If this basically works, then wire up your button and encoder logic. This way will tell you if it might be a library issue or a sketch issue.

Best of luck and let us know how it goes.

Gearhead-Ceniza, 2025-01-27:
Hi @LeeNX , thnx for your answer. 

Yeah, I've tried that and didn't work. It worked with an example Where buttons were not declared as a matrix. 
I have been trying  lots of things, diferente library versions, diferent code. The only things that worked for me was changing those parts of the library. 
I thought that declaring the Matrix should be enough, but then the software only showed 16 buttons. 

Since the serial monitor shows every button and pulse in order I think that the first part of the code  is kinds wright. But software doubles de reference every other button pressed vía bluetooth. 1-2-4-8-16-32-68 etc. 

Also, I've read in a clases issue, that libraries should be modified to show more than 16 buttons, and that is what gave me the idea of changing that part. I'm not a software guy, and understanding libraries its almost imposible for me, but I've read every fólder un the library and made sense to me. 

Thnx for ur answer. I'll take a look those links.

Gearhead-Ceniza, 2025-01-27:
Hmmm, maybe, ` #define numOfButtons 10 ` works if it substracts buttons but wont have the ability to add more than what the library declares. As soon as I arrive home I try now to add ` #define numOfButtons 20 ` with the library modified.

LeeNX, 2025-01-27:
> Hmmm, maybe, `#define numOfButtons 10` works if it substracts buttons but wont have the ability to add more than what the library declares. As soon as I arrive home I try now to add `#define numOfButtons 20` with the library modified.

The private class is setup to handle 128 on/off buttons:
```
uint8_t _buttons[16]; // 8 bits x 16 --> 128 bits
```
ie: 8bits per byte multiple by 16bytes equals 128 bits of on/off buttons

Making the change in to 64bytes, just wastes memory.

Also, making any changes to library would cause problems later, if you ever upgrade to newer release. And this should work.

When testing [DrivingControllerTest.ino]( https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/DrivingControllerTest/DrivingControllerTest.ino) sketch, this worked?

What version of library and NimBLE-Andruino were you on? Did you delete the bluetooth pairing between changes?
What gamepad testing tool are you using? I could not find anything that supports more than 16buttons. (Plus my Mac does not want to pick any new BlueTooth devices, bet it needs a Windows reboot ;-) )

Gearhead-Ceniza, 2025-01-27:
> ie: 8bits per byte multiple by 16bytes equals 128 bits of on/off buttons

Oh, now I get it. I´ll change that.

> Also, making any changes to library would cause problems later, if you ever upgrade to newer release. And this should work.

Yes, I know. But I've been trying lots of things before messing with the library.  

> When testing [DrivingControllerTest.ino](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/DrivingControllerTest/DrivingControllerTest.ino) sketch, this worked?

With that test yes, I could see those 10 buttons, but changing from 10 to 20 or so, just made the bluetooth connect and disconect and coudn't test anything.

> What version of library and NimBLE-Andruino were you on? Did you delete the bluetooth pairing between changes?
I'm using 2.2.1 version. yeah I always unpair and pair the board before testing. 

[joysticktest_092.zip](https://github.com/user-attachments/files/18564272/joysticktest_092.zip)
this is the program I'm using to test the board. 

I've just tried to add 
```c++
#define numOfButtons 28
``` 
And nothing changes. But there is no problem compiling the code.  
What I just found is that eventhough there is no warning compiling, there are 4 lines marked in red. 

> from c:\Users\name\Documents\Arduino\libraries\ESP32Encoder\src/ESP32Encoder.h:3,
                 from C:\Users\Roberto\Documents\Arduino\button-box_4-encoder_20-botones\button-box_4-encoder_20-botones.ino:1:
C:\Users\name\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.3-cfea4f7c-v1\esp32/include/driver/deprecated/driver/pcnt.h:15:2: warning: #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h" [-Wcpp]
   15 | #warning "legacy pcnt driver is deprecated, please migrate to use driver/pulse_cnt.h"

Could this be the problem, I really don´t get the meaning of this. 

Really thanx for all your replies..

LeeNX, 2025-01-28:
> Yes, I know. But I've been trying lots of things before messing with the library.

Library should make it easier. If the library is faulty (this is what I am mostly interested in), we should fix it. That and seeing what other creative people can come up with.

> With that test yes, I could see those 10 buttons, but changing from 10 to 20 or so, just made the bluetooth connect and disconect and coudn't test anything.

Would you be willing to do the brute force test? Like start at 10 buttons and do basic test, then increase by one and see that it still works, until it stops working. I am wondering if 16 is the button limit or something?

Just making sure you are deleting the pairing? Just in case unpair sounds like disconnecting vs newly paired device.

> > What version of library and NimBLE-Andruino were you on? Did you delete the bluetooth pairing between changes?
> > I'm using 2.2.1 version. yeah I always unpair and pair the board before testing.

Also using the v0.6.8 of ESP32-BLE-Gamepad? Also, did you revert the changes you made to ESP32-BLE-Gamepad?

> [joysticktest_092.zip](https://github.com/user-attachments/files/18564272/joysticktest_092.zip) this is the program I'm using to test the board.

 This is on a Windows system, have you tried with the built-in Joystick applet in the control panel? I don't have Windows setup for BLE testing, so a little harder to give ideas. Windows Joystick Applet should display the first 32 buttons states - https://github.com/MHeironimus/ArduinoJoystickLibrary/issues/100#issuecomment-652063934 

 This attachment does not say how many buttons it supports. Do you know if it support more than 16 buttons? I wonder if @lemmingDev can shed some light on how he tested the 128 button setup? Using my phone and web Testing Applets, I was only able to see 16 buttons. I tried three different web applets.
 
> #define numOfButtons 28
> And nothing changes. But there is no problem compiling the code. 
Did you delete the pairing between changes? If you VID:PID did not change, it could be that the BLE attribs are cached - [TroubleshootingGuide](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/TroubleshootingGuide.md)

> What I just found is that eventhough there is no warning compiling, there are 4 lines marked in red.
>
> Could this be the problem, I really don´t get the meaning of this.

I don't think it's a problem, more a warning. There is an issue open for that in the warning - https://github.com/madhephaestus/ESP32Encoder/issues/117

Does not look serious.

> Really thanx for all your replies..

Totally welcome, hope we making progress.  This is a super interesting problem, hope we solve it soon.

Gearhead-Ceniza, 2025-02-01:
Hi, sorry for the late reply it's been a busy week. 

> Library should make it easier. If the library is faulty (this is what I am mostly interested in), we should fix it. That and seeing what other creative people can come up with.

Now, re-reading the readme notes, it's where it says that there are 16 posible buttons and that the library can be configured up to 128. So it made total sense to me that I needed to change something and that is why I started looking for something that looked to me like a button number count/limit or similar.
 

> Would you be willing to do the brute force test? Like start at 10 buttons and do basic test, then increase by one and see that it still works, until it stops working. I am wondering if 16 is the button limit or something?

Yes, no problem, I'll try it now. 

> Also using the v0.6.8 of ESP32-BLE-Gamepad? Also, did you revert the changes you made to ESP32-BLE-Gamepad?

Yes, I had tried other versions, but finally stick with the last update.

> This is on a Windows system, have you tried with the built-in Joystick applet in the control panel? I don't have Windows setup for BLE testing, so a little harder to give ideas. Windows Joystick Applet should display the first 32 buttons states -

That is correct, windows "native" tester supports 32 buttons.

Thanks for your patience.
I'll try to get a simple piece of code and try to add button by button and test it with an unmodified library.

Gearhead-Ceniza, 2025-02-01:
I've done some tests. 
1- Using the library as is, with the default 16 buttons. The multiple button example works perfectly, will display more than 16 buttons no problem. But, using the keypad example, and changing the number of row/columns wont change the displayed buttons. I've also tried to add `#define numOfButtons 28`  and in the setup function `BleGamepadConfiguration bleGamepadConfig;
 bleGamepadConfig.setButtonCount(numOfButtons);
  bleGamepad.begin();`  . this doesn´t change the displayed buttons. It only appears the default 16 buttons.

2- Checking the keypad example, I have seen my error. I forgot to add  `bleGamepad.sendReport(); ` now the pressed buttons correspond to the displayed pressed buttons. Also the keymap should start with 1 not 0 or  the first button wont be recogniced, at least that happened to me. 

So, at this momment i've got the 20 buttons matrix working correctly but I've got to modify the library in order to have all buttons displayed. But the encoders are not working. I'll try to find out how the example from [https://github.com/MagnusThome/ESP32-BLE-Gamepad/blob/master/examples/5rotaryencoders_32buttons/5rotaryencoders_32buttons.ino](url) works. 

Thanks

lemmingDev, 2025-02-01:
Sorry

Change

```bleGamepad.begin();```

to

```bleGamepad.begin(&bleGamepadConfig);```

Gearhead-Ceniza, 2025-02-01:
> Sorry
> 
> Change
> 
> `bleGamepad.begin();`
> 
> to
> 
> `bleGamepad.begin(&bleGamepadConfig);`

Oh¡¡ I forgot to add that. It works thanks¡¡ 
As I thought, it was my fault.
I need to find a way to refresh and improve my coding abilities.

Gearhead-Ceniza, 2025-02-01:
Ok ok, now I finally got it fully working. 

Since I was using an almost direct copy of the example that @MagnusThome provided I had to understand what the code is doing.

I guessed that the rotation of the encoder, instead of looking for an incremental value it was looking for the direction, and based on the direction it would emulate a press and release of that given pulse direction. At least, that is how I think I would do it, but actually I don't really now if this is the case. 
This made me look at the Button emulation function, and its Sendkey. Looking at your examples @lemmingDev  I saw an example that tested buttons "autopressing" them.  That's when I saw that your example had a "function" ` bleGamepad.sendReport();`  that was not on the example I used. Just adding this made the encoders work. 

```c++
#include <ESP32Encoder.h>  // https://github.com/madhephaestus/ESP32Encoder/
#include <Keypad.h>        // https://github.com/Chris--A/Keypad
#include <BleGamepad.h>    // https://github.com/MagnusThome/ESP32-BLE-Gamepad

#define numOfButtons 28
BleGamepad bleGamepad("SIM_WHEEL", "gearhead", 100);

////////////////////// BUTTON MATRIX //////////////////////
#define ROWS 5
#define COLS 4
uint8_t rowPins[ROWS] = { 13, 14, 15, 16, 33 };
uint8_t colPins[COLS] = { 17, 18, 19, 21 };
uint8_t keymap[ROWS][COLS] = {
  { 1, 2, 3, 4 },
  { 5, 6, 7, 8 },
  { 9, 10, 11, 12 },
  { 13, 14, 15, 16 },
  { 17, 18, 19, 20 }
};
Keypad customKeypad = Keypad(makeKeymap(keymap), rowPins, colPins, ROWS, COLS);

//////////// ROTARY ENCODERS ////////////
#define MAXENC 4
uint8_t uppPin[MAXENC] = { 22, 25, 27, 04 };
uint8_t dwnPin[MAXENC] = { 23, 26, 32, 05 };
uint8_t encoderUpp[MAXENC] = { 21, 23, 25, 29 };
uint8_t encoderDwn[MAXENC] = { 22, 24, 26, 28 };
ESP32Encoder encoder[MAXENC];
unsigned long holdoff[MAXENC] = { 0, 0, 0, 0 };
int32_t prevenccntr[MAXENC] = { 0, 0, 0, 0 };
bool prevprs[MAXENC] = { 0, 0, 0, 0 };
#define HOLDOFFTIME 150  // TO PREVENT MULTIPLE ROTATE "CLICKS" WITH CHEAP ENCODERS WHEN ONLY ONE CLICK IS INTENDED

//////////////////////////////////////////////////////////////////////////////////////////////
void setup() {
  Serial.begin(115200);

  for (uint8_t i = 0; i < MAXENC; i++) {
    encoder[i].clearCount();
    encoder[i].attachFullQuad(dwnPin[i], uppPin[i]);
  }
  // customKeypad.addEventListener(keypadEvent);
  //customKeypad.setHoldTime(1);
  BleGamepadConfiguration bleGamepadConfig;
bleGamepadConfig.setButtonCount(numOfButtons);
  bleGamepad.begin(&bleGamepadConfig);
  
  Serial.println("Booted!");
}

/////////////////////////////////////////////////////////////////////////////////////////////
void loop() {
  unsigned long now = millis();

  // -- ROTARY ENCODERS : ROTATION -- //

  for (uint8_t i = 0; i < MAXENC; i++) {
    int32_t cntr = encoder[i].getCount();

    if (cntr != prevenccntr[i]) {
      if (!holdoff[i]) {
        if (cntr > prevenccntr[i]) { sendKey(encoderUpp[i]); } //this looks like its looking for the rotation direction to call the sendkey function 
        if (cntr < prevenccntr[i]) { sendKey(encoderDwn[i]); }
        holdoff[i] = now;
        if (holdoff[i] == 0) holdoff[i] = 1;  // SAFEGUARD WRAP AROUND OF millis() (WHICH IS TO 0) SINCE holdoff[i]==0 HAS A SPECIAL MEANING ABOVE
      } else if (now - holdoff[i] > HOLDOFFTIME) {
        prevenccntr[i] = encoder[i].getCount();
        holdoff[i] = 0;
      }
    }
  }
  
KeypadUpdate();
  
  delay(10);
}

/////////////////////////////////////////////////////////////////////////////////
void KeypadUpdate(){
  customKeypad.getKeys();
  for (int i = 0; i < LIST_MAX; i++){ // Scan the whole key list.      //LIST_MAX is provided by the Keypad library and gives the number of buttons of the Keypad instance
    if (customKeypad.key[i].stateChanged){ // Only find keys that have changed state.
            uint8_t keystate = customKeypad.key[i].kstate;
            if (bleGamepad.isConnected()){
                if (keystate == PRESSED){
                    bleGamepad.press(customKeypad.key[i].kchar);
                } // Press or release button based on the current state
                if (keystate == RELEASED){
                    bleGamepad.release(customKeypad.key[i].kchar);
                }
                bleGamepad.sendReport(); // Send the HID report after values for all button states are updated, and at least one button state had changed
            }
        }
    }
}

/////////////////////////////////////////////////////////////////////////////////
void sendKey(uint8_t key) {
  uint32_t gamepadbutton = key;
  Serial.print("pulse\t");
  Serial.println(key);
  if (bleGamepad.isConnected()) {                           //I guess this just looks for a correct bluetooth connection 
    //Serial.println("Calling bleGamepad.press()");
    bleGamepad.press(gamepadbutton);                       //now just creates a button press and release emulation
     bleGamepad.sendReport();
    //Serial.println("Returned from bleGamepad.press()");
    delay(150);
    //Serial.println("Calling bleGamepad.release()");
    bleGamepad.release(gamepadbutton);
     bleGamepad.sendReport();
    //Serial.println("Returned from bleGamepad.release()");
  }
}
``` 

It's been tough getting to this point, but also so much fun. I think that I'll buy some books to improve/ get some codding skills, cause at the moment it almost magic to me. Any suggestions would be much appreciated, like.. c++ for dummies hehehe

Thanks for your support.

LeeNX, 2025-02-01:
Congrates @Gearhead-Ceniza on getting it working! Oh that happy feeling you get when it all finally comes right or you make some progress.

As for the code you were looking at, it's just a few years old. Why I think you might have been looking at making the changes in the library. Next project, take a look at home old the code or commits are. Very helpful if there is an upstream repo or project, that might get uodated, vs a working project that somebody got to work for their project. 

Hope to see more projects and creative ideas. Please don't be a stranger. 

If your issue has been resolve, mind closing this issue. Always welcome to open new issues, if you run into anything else.

closed 2025-02-01 by Gearhead-Ceniza

lemmingDev, 2025-02-01:
> > Sorry
> > Change
> > `bleGamepad.begin();`
> > to
> > `bleGamepad.begin(&bleGamepadConfig);`
> 
> Oh¡¡ I forgot to add that. It works thanks¡¡ As I thought, it was my fault. I need to find a way to refresh and improve my coding abilities.

Was my fault as the example was missing it

Master now has it and a new release will be done soon

Glad its working