lemmingDev/ESP32-BLE-Gamepad #263: setAxes function doesn't seem to map linearly to the axis readings on gamepad tester

Cynopolis, 2025-02-04:
# Issue
There appear to be two issues that I'm seeing with this bug.

1. Axes don't appear to have a linear mapping scheme. On axis 2, an input of 0 to the `setAxes` function shows as -1 in gamepad tester. An input of 16,384 shows as 0 on gamepad tester, and an input of -32768 shows as 1 on gamepad tester.
2. Different axes have different mappings. Setting axes 0-2 to -32,768 shows as 1 on gamepad tester, but setting axis 3 to that value shows as 0 on gamepad tester.

# Expectation
1. I'd expect the axis mapping to be something like an input of 32767 to show as 1 in gamepad tester, and input of 0 to show as 0 and an input of -327678 to show as -1.
2. I'd expect the axis mapping for all exes to be consistent.

# How to Recreate

I used this site to test the gamepad:
https://hardwaretester.com/gamepad

BLE Gamepad tester version: 0.7.1 (Latest Platformio release as of 2/4/25)

Code snippet I'm using:

```
#include <BleGamepad.h>

BleGamepad xinput{};

void setup(){
  BleGamepadConfiguration config{};
  config.setAutoReport(false);
  config.setButtonCount(4);
  config.setIncludeSlider1(false);
  config.setIncludeSlider2(false);

  xinput.begin(&config);
  int16_t test{0};
  Serial.println("Starting gamepad communicaiton");
}

int16_t test{0};

void loop()
{
    // if nothing is connected to this controller then don't process any further
    if (!xinput.isConnected())
    {
      vTaskDelay(1000);
      continue;
    }
   // test the button inputs
    xinput.press(BUTTON_1);
    xinput.release(BUTTON_2);
    xinput.release(BUTTON_3);
    xinput.press(BUTTON_4);

    xinput.setAxes(-32768, 16384, test, -32768, 16384, 0, 0, 0);
    Serial.println(test);
    xinput.sendReport();
    vTaskDelay(10);
    test += 30;
}
```

lemmingDev, 2025-02-04:
By default, axes have range 0 to 32767
Only use that range, unless you configure otherwise

The actual max range, if you configure it, is a range of -32767 to 32767

Don't try to do -32768

There is a check in code for this, but maybe I forgot it on that one axis

Cynopolis, 2025-02-04:
I'm going to mark this as closed because there isn't actually an issue here. My one suggestion is to change the argument order for the rx,ry, and rz arguments, but that's no really a big deal; I just needed to pay more attention.

Thanks for your help! I really appreciate it.

lemmingDev, 2025-02-04:
Hi

Two ways of thinking to choose from

**1st way of thinking** - Industry standard is to use X, Y for left thumbstick, Z, Rz for right thumbstick, and Rx, Ry for triggers, and so standard HID descriptors/reports that contain 2 thumb sticks, 2 triggers and 2 thumb sticks are defined in the order x, y, z, rz, rx, ry, and that is the reason I chose to have setAxes function set them in that way, as this library's HID descriptor is defined in the order (x, y, z, rz, rx, ry, trigger1, trigger2)

**2nd way of thinking** - USB HID specification, usages are numbered

30 - X
31 - Y 
32 - Z 
33 - Rx 
34 - Ry 
35 - Rz
36 - Slider

and this is the order the game tester will show them, and some people argue that setAxes should be set in this order
(x, y, z, rx, ry, rz, trigger1, trigger2)

So which order should we actually use for setAxes function? @LeeNX - your thoughts?

LeeNX, 2025-02-05:
Thanks @lemmingDev ... Mmm ...

When I first saw the change this week, I was going to open an issue, because it really triggered my OCD, as I think most programmers are going to expect the 2nd way, at least that looks way neater to me.

But now that I have seen the explanation, I am thinking option 1 is better, but maybe we should put a note in the docs and examples, about why this the choice. That way the next programmer coming in, will not open an issue about how this should be changed, because it's by design and correct.

I don't known if there is a way to make it optional with a label, so that we can get both camps happy.

lemmingDev, 2025-02-05:
So, we could add a note explaining briefly as above, and then add a new function 

setAxesOrdered(x, y, z, rx, ry, rx, trigger1, trigger2)

and leave

setAxes(x, y, z, rz, rx, ry, trigger1, trigger2)

Thoughts?

LeeNX, 2025-02-05:
I think that would cover all our bases and should keep everybody happy. This would include an updated example, right?

Thanks @lemmingDev , you are awesome!

lemmingDev, 2025-02-06:
Or introduce setAxesHIDOrdered and make change setAxes to (x, y, z, rx, ry, rx, trigger1, trigger2)

lemmingDev, 2025-02-07:
Release 7.2 released in which I caved. I'll close issue, but feel free to comment. Notes below...

Existing setWhichAxes and setAxes function parameters order set to x, y, z, rx, ry, rz as most people seem to prefer this ordering

New setHIDAxes function parameters order set to x, y, z, rz, rx, ry, for people who prefer to set axes values in the order that they exist in the HID report

New setRightThumbAndroid function parameters order set to z, rx to compliment existing setRightThumb function order z, rz which is what Windows usually expects

Examples updated to reflect

closed 2025-02-07 by lemmingDev