lemmingDev/ESP32-BLE-Gamepad #264: How Can I Give the Device a Product Name?

Discman777, 2025-02-07:
This is a very useful library, and is working well.  Many thanks.

I wonder if there’s a way to specify the Game Pad Name as seen by Windows.  For instance, in the Windows’ Game Controllers app and in various games, the device shows up as “0 axis 11 button device with hat switch” etc.  This name is apparently being defined somewhere.  Because in some of the library examples, I have seen different names showing up.

I know about the Bluetooth Manufacturer and Product IDs (via the class constructor).  The Manufacturer ID shows up in the BT Settings only, and the Product ID isn't seen anywhere.

How can I change the Controller Product Name being used **in the games**?

Thank you,
   Joe

lemmingDev, 2025-02-07:
You need to change a registry entry if you want that name changed

Custom display name by means of the Windows registry editor (Windows only)

Open the "registry editor" (regedit.exe) with user privileges (not administrator).

Navigate to the following folder (registry key):
HKEY_CURRENT_USER\System\CurrentControlSet\Control\MediaProperties\PrivateProperties\Joystick\OEM

Inside that folder, locate the subfolder (registry key) for your device.

Those keys are named using this pattern: "VID -hex number- &PID -hex number- ", where "VID" means "vendor identifier" and "PID" means "product identifier". Together, they are a (plug-and-play) hardware identifier (ID). If your device uses BLE connectivity and the default hardware ID set in this project, that registry key should be "VID_E502&PID_BBAB".

Double-click (edit) the value "OEMName".

Set a custom display name and hit "enter".

NOTE: Info taken from [here](https://github.com/afpineda/OpenSourceSimWheelESP32/blob/main/doc/RenameDeviceWin_en.md#why-all-this-mess)

Discman777, 2025-02-07:
Amazing, that did the job!

So, the default OEM name was auto-generated by Windows as a stand-in?

Thanks!

lemmingDev, 2025-02-07:
> Amazing, that did the job!
> 
> So, the default OEM name was auto-generated by Windows as a stand-in?
> 
> Thanks!
Happy to help

Yeah - name is autogenerated upon first appearance of that VID/PID as a game controller

I just added the instructions to TroubleshootingGuide.md for future users

closed 2025-02-07 by lemmingDev

Discman777, 2025-02-13:
I hope you can answer another question:

I’d like to have more than 128 buttons in my application.  I’ve tried it with 192 buttons and BleGamePad appears to handle them just fine.  I had to modified BleGamepad.h to **_buttons[24]**.  My questions:

  1)  Is it safe to do that?
  2) Is there any other way to expand the capacity without modifying the .h file locally. 
 Because if I distribute my app and someone else downloads BleGamepad (from GitHub) to rebuild the code, they would need to manually modify the _buttons[].

Thanks very much.

LeeNX, 2025-02-13:
> I hope you can answer another question:

  I might not have all the answers, but I can possible give you some info, but as a side note, might have been better to open another issue, as this one was closed and might not get some many eyes.

> I’d like to have more than 128 buttons in my application. I’ve tried it with 192 buttons and BleGamePad appears to handle them just fine. I had to modified BleGamepad.h to **_buttons[24]**. My questions:

Interesting idea and has me wondering what the limitation of BLE might be. Is there a doc or something outlining the spec limitation.

> 1. Is it safe to do that, all the way up to 256 buttons?

  Not something I would do but, I am interested in the why and what is the max too. 

> 2. Is there any other way to expand the capacity without modifying the .h file locally.>    Because if I distribute my app and someone else downloads BleGamepad (from GitHub) to rebuild the code, they would need to manually modify the _buttons[].

To get around the header change, could easily note in the source what should be changed in the library. Another idea, could be distribute your projects as an PlatformIO project, which includes the modified library, but this would not get updates with new releases of ```ESP32-BLE-Gamepad``` and would be up to the user to keep track of. Still an option. Mmm, I wonder if there are limit checks and fails in the ```ESP32-BLE-Gamepad``` library, I have not checked myself.

As for more than 8 buttons, I start wondering if it would not be easier to just setup a BLE keyboard? Even then, keyboards are around 100 odd buttons. Either way, you would need to check what limitation your target Host OS might support. ie: Linux might support it, but OSX or Windows might have different implementations. If anybody finds docs or info on any of the Host OS support, that would be nice to collect and possible point too.

Last note, how would you wire up all these buttons with less GPIOs than an ESP32 device has, possible some sort of I2C GPIO multiplexer? How much lag/latency would that introduce? Something I would like to investigate, but currently have a few more other priorities I wish to get through first, before hitting this ToDo.

> Thanks very much.

 You welcome, hope we get to see your project go live and we can play with it too.

lemmingDev, 2025-02-13:
The maximum number of buttons supported in a HID descriptor depends on how the descriptor is structured and the report size.

**Standard Button Limit (Gamepads/Joysticks)**
The HID report typically defines buttons as a bit array, where each bit represents an individual button state (pressed/released).
Since a standard HID report byte is 8 bits, and you can define multiple bytes for button states, the theoretical maximum is often 128 buttons (16 bytes × 8 bits per byte).
However, many controllers use 32 or 64 buttons as a practical limit.

**Absolute Maximum (Based on Report Size)**
The maximum report size is 64 bytes for USB Full-Speed devices and 512 bytes for USB High-Speed devices.
If you only use button states (1 bit per button), a full 512-byte report could hold 4096 buttons (512 × 8).
But most firmware implementations and operating systems do not support such an extreme number.

**Real-World Limits**
- Windows DirectInput supports up to 128 buttons.
- XInput (for Xbox controllers) is limited to 16 buttons.
- Linux and macOS typically support up to 64 buttons by default.
- Gamepad APIs like SDL2 allow configurations beyond 128 buttons with custom drivers.

Discman777, 2025-02-13:
Thanks very much for your reply and information.  The device that I’m developing has a touchscreen which supports “soft pushbuttons”.  Thus, there is no real limit on how many “buttons” it can tackle.

I’ve since done a few experiments.  Here are some observations:

1) When programmed to 256 buttons, BLE Gamepad can still generate the button codes.  A controller tester app is able to display the codes.  However, Windows does not appear to recognize the device anymore.
2) Individual games also set their own upper limits as to the number of buttons they support.  The games of interest here, MSFS and X-Plane, they recognize up to 128 and 160 buttons respectively, from a controller.

So, based on the above, there’s really no point for me to pursue beyond 128 buttons.

Hope you’ll find that informative.

LeeNX, 2025-02-14:
This is all gold, need to add to a tech doc or something. See what I can do soon.