lemmingDev/ESP32-BLE-Gamepad #265: [Host bug/limitation] OSX does not list devices with more than 18 characters in device name?

LeeNX, 2025-02-09:
Seem to have discovered or tracked down by accident, that OSX seems to have a limitation of 18 characters in the device name, at least one of my running Mac with macOS version 15.3 (24D60).  
  
I really did think I was going nuts, but testing one of the example sketches - ```CharacteristicsConfiguration.ino```, using the device name ```Custom Contoller Name```, would not show up in my device list on the OSX machine, but was listed in ```LightBlue``` on my Android device.
  
Testing by changing and uploading a small difference in the sketch, turning bluetooth on the device off for 30sec and then on and waiting for 30sec to see if the device would get listed. Also validating I could see the device in ```LightBlue``` on my Android device.

I ended up at 18 characters - ```Custom-Contoller-N``` as can be seen in the screenshot.

[image]

I will try and add a note in the ```TroubleshootingGuide``` and push a PR.

lemmingDev, 2025-02-09:
Good find!

LeeNX, 2025-03-04:
This is actually a limitation of the Bluetooth Advertising packet size of 31 bytes - [Bluetooth Advertising Packet](https://novelbits.io/maximum-data-bluetooth-advertising-packet-ble/#:~:text=It%20allows%20for%20up%20to,the%20scan%20response%20packet%20payload)
  
Less ```3 bytes are occupied by the mandatory advertisement flags data.```, and then less bytes per advertising feature, like Advertised TX Power and/or Manufacturing data and so on.  
  
The display name is for initial discovery and pairing, then most hosts use the characteristics Appearance with UUID 0x2a01, I think.