lemmingDev/ESP32-BLE-Gamepad #267: Connecting to linux

Nostik-27, 2025-02-18:
when attempting to connect the esp to linux, it is not recognised as a gamepad. Works perfectly on windows.

lemmingDev, 2025-02-18:
I don't have access to a Ubuntu machine, or any other Linux environment at the moment

If you provide more information, perhaps @LeeNX might be able to weigh in

Nostik-27, 2025-02-19:
Thanks for the reply! 
Essentially I have set it up in much the same way as the examples and it works perfectly on windows. It shows in Bluetooth settings as the name I called it, and recognises it as a controller. I set it up through steam to play games and can use it to control a robot I’m working on. The problem is that when I try and connect to my Ubuntu machine, it shows in the Bluetooth manager but not as a gamepad or joystick. Instead it simply calls it an unknown device. I have it disconnected from the windows machine and the sketch on the esp is exactly the same.

Nostik-27, 2025-02-19:
update: The ubuntu machine recognises the ESP as a joypad but when it connects it suddenly sees it as an unknown bluetooth device.

LeeNX, 2025-02-20:
When I tested last and was collecting notes, I was playing on RetroPie v4.7, I think it was.  
  
I need to setup an Ubuntu box for testing, because I need to find out if all the new features are supported and if not in place, how and where to get updated.  
  
@Nostik-27 What version Ubuntu and which CPU platform, thou I think that should be less of an influence.  Maybe which Desktop environment? The bluetooth helper might not what device it is.  
  
If you up for some command line info collection, open a terminal and let's see what input devices you have.

following should list your input devices
```ls -la /dev/input/```
would look like
```
leet@odin:~$ ls -la /dev/input/
total 0
drwxr-xr-x  3 root root     100 Feb  2 14:17 .
drwxr-xr-x 18 root root    4100 Feb 20 05:59 ..
drwxr-xr-x  2 root root      80 Feb  2 14:17 by-path
crw-rw----  1 root input 13, 64 Feb  2 14:17 event0
crw-rw----  1 root input 13, 65 Feb  2 14:17 event1
```

if I remember correctly, you should see something like ```/dev/input/js0```, which we could test further.

Another idea, have you tried the [Browser Based Gamepad Test](https://hardwaretester.com/gamepad) using Chrome? I don't think FireFox has the webHID stuff enabled.

Some very old notes, but might not be helpful today, but maybe give you some ideas - https://github.com/LeeNX/ESP32-BLE-Gamepad/tree/docs-rpi#troubleshooting

Nostik-27, 2025-02-21:
It doesnt even recognise it as an input device. I'm using ubuntu 20 on an NVIDIA jetson.

LeeNX, 2025-02-23:
What version of BlueZ do you have installed? I can't find which kernel version that the Jetson has installed by default, could you also possible provide the kernel version, thou I think it's more BlueZ that does the device to input.  
  
I will try and get an older Raspberry Pi up with older Debian and see if there is anything else I can suggest.

Nostik-27, 2025-02-24:
The Kernel is 5.15.136-tegra and the version of bluez is 5.64

LeeNX, 2025-02-24:
I would say there is a bug in your ```BlueZ```, but did a quick search and run into https://forums.developer.nvidia.com/t/bluetooth-profile-accept-failed/310216
  
I don't know how big a problem it is to upgrade? The next idea I have is a self-compile of BlueZ, but that could be a big ask.

LeeNX, 2025-02-24:
Have you tried [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID)?

Nostik-27, 2025-02-24:
Yes, and unfortunately the same outcome.

lemmingDev, 2025-02-24:
What about an earlier version of this library before NimBLE was used as the default?

The final official version before that is here - v3.1:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/releases/tag/v0.3.1

A later non-official 5.1 version is here:
https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/111#issuecomment-1254260699

Check the corresponding example folder as things have changed since those releases

LeeNX, 2025-02-25:
A path forward, would be to test if any Bluetooth devices actually work. Then check if BLE devices possible work. Depending on your requirements, maybe the a project base off [ESP32-BLE-Keyboard](https://github.com/T-vK/ESP32-BLE-Keyboard) might work?

If you really up for testing, possible a project base on [ESP32-NimBLE-Keyboard](https://github.com/wakwak-koba/ESP32-NimBLE-Keyboard), thou it's pretty old.

Please let us know how it goes. Best of luck.

LeeNX, 2025-02-26:
I was going to suggest rolling back to an older deb, seen that helped something people, but when I looked up the BlueZ package for Ubuntu 20.04, seems that last package was for 5.53 - https://code.launchpad.net/ubuntu/focal/arm64/bluez

I am guessing the Jetson distro has newer packages and would not be an easy thing to roll back.  
  
I am out of ideas.

Nostik-27, 2025-02-27:
I’m away for a few days now, but I will get back to trouble shooting next week and let you know how it goes. Thanks for all your help!

lemmingDev, 2025-04-06:
> I’m away for a few days now, but I will get back to trouble shooting next week and let you know how it goes. Thanks for all your help!

Any update?

lemmingDev, 2026-09-21:
Might be worth checking out the new v0.8.0 and try the new XInput or SInput modes. They've probably got more chance of being recognised correctly