lemmingDev/ESP32-BLE-Gamepad #268: Compability with Overcooked 2.

Zn10plays, 2025-02-19:
# Describe your issue
I want to play overcooked 2 with a gamepad I made with this lib, however, no activity on the gamepad is recognized in overcooked.

I have made sure everything works by testing it on https://hardwaretester.com/gamepad.

I am using the epic games version of overcooked and all other functionality of the controller seems to be working.

I was wondering what is the cause of this, how can I make it work in my situation.

lemmingDev, 2025-02-19:
The game might need an xinput controller

This fork can work as an xinput controller
https://github.com/Mystfit/ESP32-BLE-CompositeHID

See the [XboxXInputController](https://github.com/Mystfit/ESP32-BLE-CompositeHID/tree/master/examples/GamepadExamples/XboxXInputController) example

lemmingDev, 2025-02-19:
Or you could try to launch the game via Steam

Add the Epic Games Launcher as a non-Steam game in your Steam library. 
This allows you to utilize Steam's controller configuration features for games launched through Epic.

Untested by myself

Zn10plays, 2025-02-20:
Apparently still does not work, the behavior is still the same.

Epic apparently does not have controller support on steam so path 2 also doesn't seem to work. Still looking into it right now.

lemmingDev, 2025-02-20:
https://github.com/csutorasa/XOutput will convert to controls to xinput

or

https://joytokey.net/en/ will concert controls to keyboard strokes

lemmingDev, 2025-04-01:
Any update?

Do you still need help?

lemmingDev, 2025-04-06:
Feel free to add more comments here if you still need help

closed 2025-04-06 by lemmingDev