lemmingDev/ESP32-BLE-Gamepad #269: Gamepad Blinking/Spam Button

Alsainz1, 2025-02-19:
I need help, everytime i put the code, it just spams or flicker the A button (i use the single button example) and when i press, i detects it perfectly.

LeeNX, 2025-02-20:
Can you provide copy of code you having problems with? Could be a debounce issue, have you tired [SingleButtonDebounce](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/SingleButtonDebounce/SingleButtonDebounce.ino)

lemmingDev, 2025-04-01:
Any update?

Do you still need help?

Alsainz1, 2025-04-06:
Hello, i am so sorry for not replying, was busy and i thought nobody noticed the post, it was the ESP32 probaly, for some reason it worked perfectly but then the problem did come back again, i got tired so i gave up. Maybe gonna try again with a better ESP32. Thanks LeeNX and Lemming for trying to help.

closed 2025-04-06 by Alsainz1