lemmingDev/ESP32-BLE-Gamepad #271: On Client Power Loss or ESP32 Reset, BLE-Gamepad Stuck in False Connected Mode

1stage, 2025-02-26:
When our Client device (using an ESP32-S3-WROOM) is turned off or the ESP32 reset, if our ESP32-BLE-Gamepad Server device has connected to it, no onDisconnect event gets sent, due to the abrupt power loss on the Client. The ESP32-BLE-Gamepad still shows as connected when the stats are polled...

```
21:49:59.867 -> BTLE Stats:
21:49:59.867 ->      StringAddress : 34:85:18:61:de:06
21:49:59.867 ->      NimBLEAddress : 34:85:18:61:de:06
21:49:59.912 ->       Peer Address : 34:85:18:61:de:06
21:49:59.912 ->    Peer ID Address : 34:85:18:61:de:06
21:49:59.912 ->    Peer ConnHandle : 1
21:49:59.912 ->  Peer ConnInterval : 7
21:49:59.912 ->   Peer ConnTimeOut : 600
21:49:59.912 ->   Peer ConnLatency : 0
21:49:59.912 ->           Peer MTU : 23
```

...yet there is no REAL connection. If the Client device is turned back on, it cannot re-connect with the ESP32-BLE-Gamepad, because it is already "connected" and not advertising its availability.

A power-cycle of the ESP32-BLE-Gamepad device fixes the issue... both devices re-connect with each other after both are power-cycled.

But is there a BLE-Gamepad function I can use to periodically validate the connection, and get it unstuck?

1stage, 2025-02-26:
The Client device is using NimBLE as well, but not the Arduino build.

1stage, 2025-04-01:
@lemmingDev can you shed any light on this?

LeeNX, 2025-04-25:
Mmm, I don't quite understand this issue. If the ESP32 is reset (power loss or something else), the ESP32 should reboot into a clean state, normally not been paired and would be waiting for a pairing connection. If the host OS still thinks the BLE Gamepad is still connected and is waiting for a timeout before making the stale connection disconnect. This seems more like something that the host OS needs to re-validate and I think the timeout should be investigated from the host OS.  
  
Not sure where the stats from the issue come from, guessing these are on the host OS and the times are in seconds, maybe you need to wait the 600secs (10minutes) or reduce this timeout some how. I don't think these are controlled by the ESP32-BLE-Gamepad library?

Wanting to understand this, see if I run into this problem and could possible help.