lemmingDev/ESP32-BLE-Gamepad #275: How to configure too many buttons

LittleWhite-TT, 2025-03-20:
I noticed that the readme document mentioned that 16 buttons are turned on by default. I am a rookie in this area and just an ordinary gamer. Recently, I want to diy a European Truck Simulator 2 central control box, but I don't know how to proceed. I found that this function can be achieved using esp32 through searching for information. My need is to define 30 buttons, but I don't know how to do it. If it's convenient for you, please reply me, can I? Thank you

LeeNX, 2025-03-20:
Are you looking for coding examples, like [TestAll.ino](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/TestAll/TestAll.ino), which defines almost all buttons or you looking for wiring diagram and tutor or is there something else you looking for?  
  
ESP32 has a limited number of GPIO, directly running more than 12 push buttons without some sort of analog matrix orr multiplexing device could be possible, but channeling. One of my testing MVP [Gamepad MVP](https://wokwi.com/projects/424305759534985217). 
  
Adding a mux to your project adds some other limitation that you should understand - https://www.instructables.com/Tutorial-74HC4067-16-Channel-Analog-Multiplexer-De/ explains quite nicely.  
  
Let us know how it goes with your project.

LittleWhite-TT, 2025-03-20:
Thank you for your answer. I think I need to learn knowledge about electronic circuits and programming to understand what you mean. Thank you again

LeeNX, 2025-03-20:
You are welcome. https://wokwi.com/ is a great place to play around with circuits and programming, just does not support testing Bluetooth.  
  
If there is nothing else we can help you with, minding closing the issue. Thanks.

closed 2025-03-22 by LittleWhite-TT