lemmingDev/ESP32-BLE-Gamepad #276: Limited to 7 buttons

MatoXD99, 2025-03-27:
Hi. I can't get the device past 7 buttons. It either wont broadcast BLE signal anymore or if it does work, it limits it to 7 buttons (for an example in Asseto Corsa). I tried also with most of the basic examples that come with the library, but it still limits to 7 buttons, can't get past that. If I open the device in windows settings, I also get:

"Test the game controller. If the controller is not functioning properly, it may need to be calibrated. To calibrate it, go to the Settings page".

Is this a bug in the library, Arduino IDE, ...? I'm using ESP32 C3.

LeeNX, 2025-03-27:
Without a sketch/code and/or wire diagram, here a bunch of things you could try.

Have you tried any of the suggestions in the [Troubleshooting Guide](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/TroubleshootingGuide.md)?  

Unpairing and deleting the Bluetooth devices is a must between changes, as suggested in the Troubleshooting Guide.

Have you tried to used some sort of BLE Scanner tool or app and see what your device is reporting? I like to add something like a battery changing attrib, so that I can see basic BLE is working - [bleGame-Tester.ino](https://gitlab.com/leet/ble-gamepad-collection/-/blob/main/arduino/bleGame-Tester/bleGame-Tester.ino?ref_type=heads), thou this sketch does not have any buttons, you should see changes happening in Host OS (if it supports battery reporting) and BLE Scanner. Make user you set the LED pin correctly, as this has bitten me in the you know where, which I have switched between the original ESP32 and ESP32-C3 (generally on pin 8 vs pin 2 for the original, but check your docs).

There should be no basic problems with any of the Examples, like the sketch [TestAll.ino](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/TestAll/TestAll.ino) which tests 64 buttons triggered to the Host OS, thou no really physical buttons are wired up.  

Don't believe there should be any known bugs in either the ESP32-BLE-Gamepad Library or dependancy [NimBLE-Arduino](https://github.com/h2zero/NimBLE-Arduino) or the Example sketches. Many people, including myself have tested these sketches unmodified.

I hope this helps and you can make progress with your project. Let us know how it goes. Best of luck.

lemmingDev, 2025-03-27:
You're probably using a GPIO you shouldn't be...

MatoXD99, 2025-03-28:
The problem appearently was me not unpairing and re-pairing each time I changed button numbers... Kinda weird but it works. Thanks @LeeNX

closed 2025-03-28 by MatoXD99

LeeNX, 2025-03-28:
@MatoXD99 glade you found the problem. One would guess that if you just changing the internal of the sketch, you should not need to re-pair, because the HID does not change, but been caught too many times and not really compared the HID between changes. Just easier to unpair and pair again.  
  
Best of luck with your project, hope you will post links to what you doing. Always interested in seeing what other people are doing.