lemmingDev/ESP32-BLE-Gamepad #277: Is it possible to disable joystick

mdshoaib2002, 2025-03-30:
Tried many things but not able to disable joystic function 
i just wanted hat switch and 8 buttons all are working fine 
just joystick is pointing up-left constatnly 
any solution for this..??

1stage, 2025-03-30:
What code are you using? One of the examples? Post it here, please.

mdshoaib2002, 2025-03-31:
#include <BleGamepad.h>

BleGamepad bleGamepad;

#define UP_PIN    23
#define DOWN_PIN  22
#define LEFT_PIN  16
#define RIGHT_PIN 17

// Button GPIOs
#define X_BUTTON      25
#define SQUARE_BUTTON 26
#define TRIANGLE_BUTTON 27
#define CIRCLE_BUTTON 32
#define R1_BUTTON  33
#define L1_BUTTON  18
#define START_BUTTON 21
#define SELECT_BUTTON 19

void setup() {
    pinMode(UP_PIN, INPUT_PULLUP);
    pinMode(DOWN_PIN, INPUT_PULLUP);
    pinMode(LEFT_PIN, INPUT_PULLUP);
    pinMode(RIGHT_PIN, INPUT_PULLUP);

    pinMode(X_BUTTON, INPUT_PULLUP);
    pinMode(SQUARE_BUTTON, INPUT_PULLUP);
    pinMode(TRIANGLE_BUTTON, INPUT_PULLUP);
    pinMode(CIRCLE_BUTTON, INPUT_PULLUP);
    pinMode(R1_BUTTON, INPUT_PULLUP);
    pinMode(L1_BUTTON, INPUT_PULLUP);
    pinMode(START_BUTTON, INPUT_PULLUP);
    pinMode(SELECT_BUTTON, INPUT_PULLUP);

    bleGamepad.begin();
}

void loop() {
    if (bleGamepad.isConnected()) {
        uint8_t hatSwitch = 0;  // Default: Center (no direction)

        bool up = digitalRead(UP_PIN) == LOW;
        bool down = digitalRead(DOWN_PIN) == LOW;
        bool left = digitalRead(LEFT_PIN) == LOW;
        bool right = digitalRead(RIGHT_PIN) == LOW;

        if (up) hatSwitch = 1;
        if (down) hatSwitch = 5;
        if (left) hatSwitch = 7;
        if (right) hatSwitch = 3;
        if (!up && !down && !left && !right) hatSwitch = 0;

        bleGamepad.setHat(hatSwitch);

        // Handle buttons
        if (digitalRead(X_BUTTON) == LOW) bleGamepad.press(BUTTON_1); else bleGamepad.release(BUTTON_1);
        if (digitalRead(SQUARE_BUTTON) == LOW) bleGamepad.press(BUTTON_2); else bleGamepad.release(BUTTON_2);
        if (digitalRead(TRIANGLE_BUTTON) == LOW) bleGamepad.press(BUTTON_3); else bleGamepad.release(BUTTON_3);
        if (digitalRead(CIRCLE_BUTTON) == LOW) bleGamepad.press(BUTTON_4); else bleGamepad.release(BUTTON_4);
        if (digitalRead(R1_BUTTON) == LOW) bleGamepad.press(BUTTON_5); else bleGamepad.release(BUTTON_5);
        if (digitalRead(L1_BUTTON) == LOW) bleGamepad.press(BUTTON_6); else bleGamepad.release(BUTTON_6);
        if (digitalRead(START_BUTTON) == LOW) bleGamepad.press(BUTTON_9); else bleGamepad.release(BUTTON_9);
        if (digitalRead(SELECT_BUTTON) == LOW) bleGamepad.press(BUTTON_10); else bleGamepad.release(BUTTON_10);
    }

    delay(10);

lemmingDev, 2025-03-31:
Doesn't really matter I guess

Anyway - if it's working apart from that, this will fix your code

```
#include <BleGamepad.h>

BleGamepad bleGamepad;

#define UP_PIN    23
#define DOWN_PIN  22
#define LEFT_PIN  16
#define RIGHT_PIN 17

// Button GPIOs
#define X_BUTTON      25
#define SQUARE_BUTTON 26
#define TRIANGLE_BUTTON 27
#define CIRCLE_BUTTON 32
#define R1_BUTTON  33
#define L1_BUTTON  18
#define START_BUTTON 21
#define SELECT_BUTTON 19

void setup() {
    pinMode(UP_PIN, INPUT_PULLUP);
    pinMode(DOWN_PIN, INPUT_PULLUP);
    pinMode(LEFT_PIN, INPUT_PULLUP);
    pinMode(RIGHT_PIN, INPUT_PULLUP);

    pinMode(X_BUTTON, INPUT_PULLUP);
    pinMode(SQUARE_BUTTON, INPUT_PULLUP);
    pinMode(TRIANGLE_BUTTON, INPUT_PULLUP);
    pinMode(CIRCLE_BUTTON, INPUT_PULLUP);
    pinMode(R1_BUTTON, INPUT_PULLUP);
    pinMode(L1_BUTTON, INPUT_PULLUP);
    pinMode(START_BUTTON, INPUT_PULLUP);
    pinMode(SELECT_BUTTON, INPUT_PULLUP);

    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setWhichAxes(false, false, false, false, false, false, false, false);  
    bleGamepad.begin(&bleGamepadConfig);
}

void loop() {
    if (bleGamepad.isConnected()) {
        uint8_t hatSwitch = 0;  // Default: Center (no direction)

        bool up = digitalRead(UP_PIN) == LOW;
        bool down = digitalRead(DOWN_PIN) == LOW;
        bool left = digitalRead(LEFT_PIN) == LOW;
        bool right = digitalRead(RIGHT_PIN) == LOW;

        if (up) hatSwitch = 1;
        if (down) hatSwitch = 5;
        if (left) hatSwitch = 7;
        if (right) hatSwitch = 3;
        if (!up && !down && !left && !right) hatSwitch = 0;

        bleGamepad.setHat(hatSwitch);

        // Handle buttons
        if (digitalRead(X_BUTTON) == LOW) bleGamepad.press(BUTTON_1); else bleGamepad.release(BUTTON_1);
        if (digitalRead(SQUARE_BUTTON) == LOW) bleGamepad.press(BUTTON_2); else bleGamepad.release(BUTTON_2);
        if (digitalRead(TRIANGLE_BUTTON) == LOW) bleGamepad.press(BUTTON_3); else bleGamepad.release(BUTTON_3);
        if (digitalRead(CIRCLE_BUTTON) == LOW) bleGamepad.press(BUTTON_4); else bleGamepad.release(BUTTON_4);
        if (digitalRead(R1_BUTTON) == LOW) bleGamepad.press(BUTTON_5); else bleGamepad.release(BUTTON_5);
        if (digitalRead(L1_BUTTON) == LOW) bleGamepad.press(BUTTON_6); else bleGamepad.release(BUTTON_6);
        if (digitalRead(START_BUTTON) == LOW) bleGamepad.press(BUTTON_9); else bleGamepad.release(BUTTON_9);
        if (digitalRead(SELECT_BUTTON) == LOW) bleGamepad.press(BUTTON_10); else bleGamepad.release(BUTTON_10);
    }

    delay(10);
```

The magic is in the last 3 lines of the setup function, where it disables all possible 8 analog axes
```
    BleGamepadConfiguration bleGamepadConfig;
    bleGamepadConfig.setWhichAxes(false, false, false, false, false, false, false, false);  
    bleGamepad.begin(&bleGamepadConfig);
```

Please test and respond

lemmingDev, 2025-03-31:
And remember, because the HID descriptor will be changed, you need to remove/delete the bluetooth device from the system and then search for and pair it again

mdshoaib2002, 2025-03-31:
Thanks this worked 
Anyhow i was wondering can any further changes will able this device connect to a PS4 via blutooth ...???

LeeNX, 2025-04-01:
From my limited searching, PS4 does not support many other bluetooth devices that are not Sony certified. I am guessing to reduce cheating and/or similar behaviour. I don't have a PS4 to test further.  
  
Looks like some people have made USB devices, like USB gamepads, keyboards and mice. Maybe a bluetooth to USB dongle could be made, but that falls outside of the scope of this library. Would be an interesting project. Looks like somebody has already done something like that [BlueRetro](https://github.com/darthcloud/BlueRetro) or maybe [GIMX](https://gimx.fr/wiki/index.php?title=Main_Page)
  
If this issue is resolved, would you mind closing the issue?

LeeNX, 2025-04-01:
**Rabbit hole warning!!**

Might be able to make a Bluetooth(maybe BLE) keyboard or mouse, with [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID) and pairing. Found a video that suggests keyboards and mice are possible supported - [How To Connect Keyboard and Mouse to PS4! [With / Without USB]](https://youtu.be/JpzAbhU5Zjg?si=5UzoCjjeuXJh9e8f). I am not sure if you would be able to bind set keys or mice input into all games or applications on the PS4, but should be pretty easy to do a keyboard with just the cursor keys and a few other buttons. 
  
As always, BLE Gamepad support is up to the Host OS to support.

Best of luck with your project.

lemmingDev, 2025-04-01:
https://github.com/csutorasa/XOutput will convert this existing controller to xinput

or, use [ESP32-BLE-CompositeHID](https://github.com/Mystfit/ESP32-BLE-CompositeHID) as stated above

mdshoaib2002, 2025-04-01:
can we set axes value to be centered please

LeeNX, 2025-04-02:
Change the following line
```c
bleGamepadConfig.setWhichAxes(false, false, false, false, false, false, false, false);  
```
to
```
bleGamepadConfig.setWhichAxes(true, true, false, false, false, false, false, false);  
```
This will give you a single x/y axes.

and then something like
```c
        uint8_t hatSwitch = 0;  // Default: Center (no direction)

        int16_t x = 0;  // Default x
        int16_t y = 0;  // Default y

        bool up = digitalRead(UP_PIN) == LOW;
        bool down = digitalRead(DOWN_PIN) == LOW;
        bool left = digitalRead(LEFT_PIN) == LOW;
        bool right = digitalRead(RIGHT_PIN) == LOW;

        // Directional logic (X/Y and Hat Switch)
        if (up && left) {
            hatSwitch = 7;
            x = -32767;
            y = 32767;
        } else if (up && right) {
            hatSwitch = 1;
            x = 32767;
            y = 32767;
        } else if (down && left) {
            hatSwitch = 5;
            x = -32767;
            y = -32767;
        } else if (down && right) {
            hatSwitch = 3;
            x = 32767;
            y = -32767;
        } else if (up) {
            hatSwitch = 0;
            y = 32767;
        } else if (down) {
            hatSwitch = 4;
            y = -32767;
        } else if (left) {
            hatSwitch = 6;
            x = -32767;
        } else if (right) {
            hatSwitch = 2;
            x = 32767;
        }

        bleGamepad.setHat(hatSwitch);
        bleGamepad.setAxes(x, y);

```
I have not compile or tested, but looks about right. Let us know if your device works as expected.

lemmingDev, 2025-04-06:
Good to hear - will close issue

closed 2025-04-06 by lemmingDev