lemmingDev/ESP32-BLE-Gamepad #280: No problem, just after some help

andy-lee4, 2025-04-25:
Hi, I am attempting to use the library with a XIAO ESP32C3 to convert a sim steering wheel controller with buttons and 2 analog axis. The buttons all work fine but I am at a loss for how to assign slider 1 and slider 2 to particular analog input pins. I have not been able to work it out from the examples, any info would be really appreciated.

LeeNX, 2025-04-25:
Something like the [DrivingControllerTest.ino](https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/master/examples/DrivingControllerTest/DrivingControllerTest.ino)

Around https://github.com/lemmingDev/ESP32-BLE-Gamepad/blob/33a3dee3b610da6243fb0472f02d2f80fd0ecf63/examples/DrivingControllerTest/DrivingControllerTest.ino#L89

Replace the for loop, where you read your analog pot value and replace the varible```i``` in ```bleGamepad.setAccelerator(i);```. 

Similar with ```bleGamepad.setAccelerator(i);```

A Wokwi project - [ESP32_C3_Gamepad_MVP_RGB](https://wokwi.com/projects/424305759534985217), without the BLE part, because Wokwi does not have an simulator for BLE, but you should be able to get the idea.

I really should commit my example for this Wokwi project, see what I can do this weekend.  
  
Let us know if how things go.

andy-lee4, 2025-04-25:
@LeeNX  Thanks for the reply. It was a me problem, I was reading the analog and mapping the range but had the setautoreport = false as I was using the 'MultipleButtonDebounce' example as a starting point for the button inputs. The analog values were only updating when a button input was pressed due to the 'sendReport'. I'm now setting this true when the analog values change and everythingis working as expected. Thank you.

LeeNX, 2025-04-25:
Glade it was easy to solve. Enjoy your new toy!