lemmingDev/ESP32-BLE-Gamepad #284: Driving controllerTest example doesn't work with Forza Horizon 4

avfonseca00, 2025-05-07:
Hi friends! I've been trying this library on my ESP32 Devkit C v4 to simulate a driving controller to play Forza Horizon 4 but the game doesn't recognize the control as a steering wheel. I also tested on Euro Truck Simulator 2 and it works fine. If you could find out how to make it compatible with that game I'll be very thankful to you. Anyway thanks for your awesome work.

lemmingDev, 2025-05-07:
https://github.com/csutorasa/XOutput will convert controls to xinput

or

https://joytokey.net/en/ will convert controls to keyboard strokes

or, use the xinput compatible fork of this library at 

https://github.com/Mystfit/ESP32-BLE-CompositeHID

See the [XboxXInputController](https://github.com/Mystfit/ESP32-BLE-CompositeHID/tree/master/examples/GamepadExamples/XboxXInputController) example

avfonseca00, 2025-05-07:
Thank you for the answer friend. I've trying to use the xinput compatible fork of this library you mentioned before and I can't see how to simulate the steering wheel because the xinput example is for a joystick controller and I would like to know if you have any idea of implementing it using that library version. Thanks beforehand.

lemmingDev, 2025-05-09:
I would assume the x axis is steering, left trigger is brake and right trigger is accelerator, though you should check

lemmingDev, 2025-05-09:
Yeah - what I said above looks correct, and here are some more
Note: Where steering says L, it means left thumbstick, which would mean the x axis like I said

[image]

avfonseca00, 2025-05-12:
You're right! I figured it out a few days ago (testing each control with an encoder until I got the expected result), but what I still don't understand is how to get the device to be detected as a steering wheel in the game, since this setup works like a joystick. I believe there are different settings for a steering wheel in the game that offer a better and much more realistic gaming experience.

avfonseca00, 2025-05-12:
I've been working on coding my own class for the steering wheel based on the example of the XBox joystick but I'm not sure of how to get the right device descriptor for a steering wheel

lemmingDev, 2025-05-13:
In Windows for example, the operating system uses the device VID and PID to detect known steering wheel controllers, and then a lot of them then use standard x, y, z, zR axis for steering, accelerator, brake and clutch controls

You could try using Vendor ID 0x046D (Logitech Inc.) and Product ID 0xC24F (G29 Driving Force Racing Wheel) and see if you can get it to work

avfonseca00, 2025-05-13:
Thanks, I'll try and let you know.

closed 2025-09-22 by lemmingDev