lemmingDev/ESP32-BLE-Gamepad #285: I'm very confused

Hidekixa, 2025-05-11:
I don't speak English and I don't use github either, I made the buttons send via ble, it's working on the gamepad tester, but the joystick doesn't want to work, at rest it stays at X=1872 and Y=1899 varying little, I've tried everything, but it always gets stuck up left, when I move it to the right it even goes down too, but I can't leave it in the center at rest or move to the left and up, because they "teleport" to the opposite side

this is my code

#include <BleGamepad.h>

BleGamepad bleGamepad("Batman", "Fabricante que nois é", 100);

const int buttonPins[4] = {12, 14, 27, 26};

#define VRX_PIN  34 
#define VRY_PIN  35  

#define AXIS_MIN      0
#define AXIS_MAX      4095
#define GAMEPAD_MIN  -32767
#define GAMEPAD_MAX   32767

void setup() {
  Serial.begin(115200);

  for (int i = 0; i < 4; i++) {
    pinMode(buttonPins[i], INPUT_PULLUP);
  }

 
  bleGamepad.begin();
}

void loop() {
  if (bleGamepad.isConnected()) {

  
    for (int i = 0; i < 4; i++) {
      if (digitalRead(buttonPins[i]) == LOW) {
        bleGamepad.press(i + 1);    
      } else {
        bleGamepad.release(i + 1);  
      }
    }

    int rawX = analogRead(VRX_PIN);
    int rawY = analogRead(VRY_PIN);

   
    int joyX = map(rawX, AXIS_MIN, AXIS_MAX, GAMEPAD_MIN, GAMEPAD_MAX);
    int joyY = map(rawY, AXIS_MIN, AXIS_MAX, GAMEPAD_MIN, GAMEPAD_MAX);

    bleGamepad.setX(joyX);
    bleGamepad.setY(joyY);
  }

  delay(10);
}

and this is just the code buttons

#include <BleGamepad.h>

BleGamepad bleGamepad("Batman", "Fabricante que nois é", 100);

const int buttonPins[4] = {12, 14, 27, 26};

void setup() {
  Serial.begin(115200);

  
  for (int i = 0; i < 4; i++) {
    pinMode(buttonPins[i], INPUT_PULLUP);
  }

  bleGamepad.begin();
}

void loop() {
  if (bleGamepad.isConnected()) {
    for (int i = 0; i < 4; i++) {
      if (digitalRead(buttonPins[i]) == LOW) {
        bleGamepad.press(i + 1);    
      } else {
        bleGamepad.release(i + 1); 
      }
    }
  }

  delay(10);  
}

LeeNX, 2025-05-16:
Welcome @Hidekixa , if you not using github, I am not sure how you might be getting updates on your issue request, ut let's see if I can leave some info in this issue.

A similar issue came up a few weeks back - https://github.com/lemmingDev/ESP32-BLE-Gamepad/issues/277. 
  
I have some wiring diagrams and different testing sketches in my repo - [BLE Gamepad Collection](https://gitlab.com/leet/ble-gamepad-collection)
  
And an emulated testing code (no BLE,  Wokwi don't support Bluetooth, yet) [ESP32_C3_Gamepad_MVP_RGB](https://wokwi.com/projects/424305759534985217)

I have been meaning to write and test the basic working gamepad.

LeeNX, 2026-08-27:
There has been no feedback for more than a year, I think we can close this. Welcome to re-open if you need to.

closed 2026-08-27 by LeeNX