lemmingDev/ESP32-BLE-Gamepad #294: [Feature Request] Add HID battery feedback

LeeNX, 2025-09-18:
Adding an input HID descriptor that has the battery level populated.  Means that applications that have access to the HID descriptor, can also read the battery level.

Not all applications have access to the GATT level of the device.

LeeNX, 2026-08-27:
The original idea was to report on the input HID report, I see that SInput as access via the Feature Reports, which works.

My first tests with using the input HID report, was that most hosts don't know what to do with battery level and just make it look like an analog trigger, which can still work, but I think any game or application might try and hook it into an input.

lemmingDev, 2026-09-21:
Hey

You're right, and I checked all four modes — the gap is real but uneven:

- **Generic**: no battery in the HID map at all, GATT Battery Service only. Exactly your complaint.
- **One S (1708)**: actually *declares* a battery input report (`0x04`) but never updates it — half-built, my bad.
- **Series X**: the real capture has no battery report, so nothing to wire without deviating from it. Leaving that one.
- **SInput**: already carries charge level in-report. Nothing to do.

On the "game hooks it as an input" worry — fair concern, and there are two ways to place the byte, with different risk:

- **Separate report ID (my recommendation)**: battery lives in its own report, same pattern as Microsoft's `0x04` on the 1708. The main gamepad report stays byte-identical on or off, so even the most aggressive mapper only ever sees an unchanged primary report. Slightly more code (second characteristic + pack path), all established patterns already in the codebase.
- **Appended to the main report**: simpler — one extra byte at the end, existing offsets untouched. But it grows the report length, so length-validating parsers could reject the whole report, and exhaustive mappers might surface it as assignable. Realistically most parsers key on usages (`0x06:0x20` is unambiguously not a game control — real controllers ship battery in HID maps already), but "most" isn't "all".

Either way it'd be **default off**, so nobody's existing mappings shift unless they opt in. I'm leaning separate-report; shout if you'd take the appended byte instead.

You want me to just dig in and get this done?

LeeNX, 2026-09-22:
> Hey

Yo ...

> You're right, and I checked all four modes — the gap is real but uneven:

Was just an idea while trying to get one of my RetroPies to report battery level and charging.

> On the "game hooks it as an input" worry — fair concern, and there are two ways to place the byte, with different risk:
>
> * **Separate report ID (my recommendation)**: battery lives in its own [...]

I think Appended to the main report could help apps that don't have access to GATT or an OS reporting battery service.

> Either way it'd be **default off**, so nobody's existing mappings shift unless they opt in. I'm leaning separate-report; shout if you'd take the appended byte instead.

Off by default is good, not sure if anybody else might even use it. They might need to make some changes so the slide/trigger does not bind to something in a an app/game/emulator

> You want me to just dig in and get this done?

Please go ahead, if you have time. I am AFK for the next week or so, but will try and check-in when I can.