lemmingDev/ESP32-BLE-Gamepad #295: How to save esp32 on battery consumption

Pac333, 2025-09-19:
Does anyone know how to drastically reduce battery consumption when working on this project?

In my experience, the limit was to reduce it to 40ma based on esp32-c3
To share the experience, if you lower the clock speed below 40 mhz, neither wifi/bt module works. Turning off the module didn't do much either

[image]
I heard that if you enter lightsleep as shown in the image above, you can use rtc to reduce power consumption without bluetooth interruption, but I always get bluetooth disconnection...

[](https://forum.seeedstudio.com/t/esp32c3-power-consumption/276948)
Then when I saw this person's post, I connected WIFI in advance and BT in the setup function, I actually cut my battery consumption by half to 80ma -> 40ma

BT Only
Bus Voltage: 4.82 V   Current: 85.50 mA   Power: 410.00 mW
Bus Voltage: 4.83 V   Current: 82.70 mA   Power: 400.00 mW
Bus Voltage: 4.83 V   Current: 82.40 mA   Power: 400.00 mW
Bus Voltage: 4.82 V   Current: 80.60 mA   Power: 394.00 mW
Bus Voltage: 4.82 V   Current: 88.90 mA   Power: 404.00 mW
Bus Voltage: 4.82 V   Current: 80.00 mA   Power: 390.00 mW

BT + WIFI
Bus Voltage: 4.89 V   Current: 39.90 mA   Power: 196.00 mW
Bus Voltage: 4.88 V   Current: 44.90 mA   Power: 220.00 mW
Bus Voltage: 4.89 V   Current: 44.90 mA   Power: 220.00 mW
Bus Voltage: 4.88 V   Current: 44.60 mA   Power: 220.00 mW
Bus Voltage: 4.88 V   Current: 45.00 mA   Power: 220.00 mW
Bus Voltage: 4.88 V   Current: 45.10 mA   Power: 220.00 mW

I'm using a translator, so the sentence might be a little weird🥲

LeeNX, 2025-09-20:
Something I had planned to look at, but don't yet have the equipment good enough to measure this small amount of power usage and using average usage over a period of time has too many factors for me to think it's worth working on yet, but ...  
  
The above info is very interesting. Do you have a MVP(minimum viable program) to run for testing?

50% power saving sounds is huge, but would need to really do some profiling. Things like long term in use runs vs power-on testing. Is the same seen for all ESP32 BLE devices or is this only the ESP32-C3 devices? I would also wonder about all the versions of libraries and toolchain used to build and run the testing.

As related to ```ESP32-BLE-Gamepad```, there is an option, where you can reduce the TX power setting and I think you can reduce advertising times, but I think that would only be for none active or IDLE time, like waiting between scenes in games or something. Other things that would need to be taken into account, is how much latency would be introduced if you lower things like the clock rate? Latency testing a project I am working on and hope to get something out soon or as soon as spare time permits.

Pac333, 2025-09-20:
I don't have good equipment either 🥲
I used the remaining Uno and INA219 breakout boards
In my experience, the power didn't surge in a hurry even when I pressed the button. Instead, it flowed to 40 mA, and then the One Tick (0.5 sec) soared by 6-70 mA each. It's back to 40 mA. Still, it's much less used than the traditional 80-90 mA

I use the vs code with pio, but all the libraries have been updated to the latest version. I didn't specify a specific version

```c
#include <Wire.h>
#include <Adafruit_INA219.h>

Adafruit_INA219 ina219;

void setup() 
{
  Serial.begin(115200);
  while (!Serial)
  {/*Serial Conenct Waiting*/}

  uint32_t currentFrequency;
  
  Serial.println("INA219 Start measuring current");
  
  if (!ina219.begin()) 
  {
    Serial.println("INA219 not found, please check the connection");
    while (1);
  }
}

void loop() 
{
  float bus_voltage = 0;
  float current_mA = 0;
  float power_mW = 0;

  bus_voltage = ina219.getBusVoltage_V();
  current_mA = ina219.getCurrent_mA();
  power_mW = ina219.getPower_mW();

  Serial.print("Bus Voltage: "); 
  Serial.print(bus_voltage); 
  Serial.print(" V   ");

  Serial.print("Current: "); 
  Serial.print(current_mA); 
  Serial.print(" mA   ");

  Serial.print("Power: "); 
  Serial.print(power_mW); 
  Serial.println(" mW");

  delay(500);
}
```
[image]
This is the code and wiring uploaded to Uno

```c
#include <Arduino.h>
#include <BleGamepad.h>
#include <WiFi.h>

// Wifi Configuration
const char* ssid = "WIFI SSID";       // wifi name, dont remove " "
const char* password = "WIFI PW"; // wifi pw, dont remove " "

BleGamepad bleGamepad;

void setup()
{
    Serial.begin(115200);
    Serial.println("Start BLE");
    bleGamepad.begin();
    
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) 
    {
        delay(500);
        Serial.print(".");
    }
}

void loop()
{
    if (bleGamepad.isConnected())
    {
        bleGamepad.press(BUTTON_5);
        bleGamepad.press(BUTTON_16);
        bleGamepad.pressStart();
        bleGamepad.setAxes(32767, 32767, 32767, 32767, 32767, 32767, 32767, 32767);
        bleGamepad.setHat1(HAT_DOWN_RIGHT);
        delay(500);

        bleGamepad.release(BUTTON_5);
        bleGamepad.releaseStart();
        bleGamepad.setHat1(HAT_CENTERED);
        bleGamepad.setAxes(0, 0, 0, 0, 0, 0, 0, 0);
        delay(500);
    }
}
```
It's the test code of esp32c3, which only adds wifi to the example code provided as standard. I tested it with this, and the actual gamepad in use had similar power consumption to this example code
Uno's serial monitor will show usage

I only have a lot of esp32c3 and I don't have any other boards, so I don't know if it's a problem with the whole esp32, but after looking for a long time to save the battery, there were a lot of people who said the whole esp32 had a ridiculous battery consumption when connecting to the BLE
I'm not sure...🥲

LeeNX, 2025-09-21:
Nice write up @Pac333 

Some further ideas, not that I have tried any of them myself, but have you looked at changing the Transmit power levels with ```bleGamepadConfig.setTXPowerLevel(txPowerLevel);  // Defaults to 9 if not set. (Range: -12 to 9 dBm)```. 
  
This would effect how far away your Host Device is. I would also wonder how the WiFi signal would effect power usage, if the Access Point was further away and the ESP32 device was searching for another connection.  
  
If I remember all the reading and I could be so wrong, but the WiFi and BlueTooth components share the same radio and would not think that having all the extra resources running for WiFi would save power in any way. Plus I have seen projects with a little more complex sketch run into resource issues, like running out of RAM.

Funny that the BLE been BlueTooth ```Low Energy``` seems to be consuming more power than expected. I would think the ESP32-C3 been so small and only has BLE, no BlueTooth classic, would also be a power saver. Also could be the Arduino Framework that is a little less optimised. Another idea would be to test using Platform.IO, where you can select different frameworks and possibly even different versions of the frameworks. 

Platform.IO gives you a little more control over your environment per project, like https://gitlab.com/leet/ble-gamepad-collection/-/blob/main/platformio/esp32blegamepad-output-rumble/platformio.ini?ref_type=heads which is one of the testing projects (not so successful yet), but enabling debug and library version selection, but that I have messed with framework variations. 

Outside of gamepad project testing, I normally use ESPHome, which has some pretty crazy build options.  
  
Let's us know what you find, maybe somebody else might have more info to add.

Thanks